sidoh/esp8266_milight_hub #721: nRF24L01+PA+LNA Antenna

JonnyD136, 2021-04-08:
Hi,
I cant get nRF24L01+PA+LNA Antenna to work. nRF24L01 works fine. Can anybody help me or is there another problem with this RF Module?

gzxto1, 2021-04-09:
Using a nodemcuV3 I just had to unplug he old nRF24L01 and plugin the new nRF24L01+PA+LNA. Worked flawless..

JonnyD136, 2021-04-09:
I‘ve tried it again but it doesn’t work. I‘m using a nodemcuV3 too.

sidoh, 2021-04-09:
Hello,

I've had a fair amount of trouble with this module as well (and honestly, nrf24s in general). I've never felt like I've managed to pinpoint what's going on, but users have reported success with some combination of:

1. Adding a capacitor across vin & gnd
2. Ensuring you're using a quality power supply
3. Making extra sure connections are tight (ideally socketed or soldered)
4. Ensuring distance from the nRF24 and ESP8266 antennae (they interfere with each other)

AlexKusnezov, 2021-04-13:
just like @gzxto1 mentioned, I've also just replaced the NRF24L01 with NRF24L01+PA+LNA and it worked right away with no issue:
[image]

btw, rocking like this for almost 2 years now and it has been the most sable device in my setup so far. 3x FUT014, 8x FUT103, 6x fut011 and 1x Fut013. wife's happy :)

JonnyD136, 2021-04-13:
Thank you guys! I will try the capacitor and make some distance between them. I´ve tried to pair with a GU10 Bulb and it worked but no control possible. Waiting for the capacitor and Antenna Cable now.

AlexKusnezov, 2021-04-13:
@JonnyD136 maybe you just need some performance tuning, check this out:
https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning
I'm using this anetnna:
https://www.amazon.de/gp/product/B06Y43D6MC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

JonnyD136, 2021-04-26:
Hello,
a capacitor solved the problem for me! Now it works with small and big Antenna.
Thank you guys!