sidoh/esp8266_milight_hub #722: LT8920 Wiring Help

fluffymadness, 2021-04-16:
I bought the following module from aliexpress https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.buyer_waiting_review.0.0.46ce6c1boDd7HN&orderId=8128714134058918&productId=32917687836
because I've read in a git issue lt8900 and lt8920 should be more performant
since the nrf has to emulate the protocol via software.
However, I'm not really sure how to wire this one, since it has different pinout
then the lt8900 description and the nrf one.
Maybe someone can point me in the right direction.
Thanks.

sidoh, 2021-04-23:
Hey @fluffymadness,

I would not recommend using an LT8900. The support for it isn't quite complete. In particular, listening/sniffing doesn't work.

For the pinout, I think SS, CLK, MOSI, MISO are all standard SPI pins. Connect them to the appropriate SPI pins on the ESP8266. The non-SPI pins are PKT and RESET. These are both configurable in the "Setup" tab of settings, so connect them to whatever pin you wish (D1/GPIO5, D2/GPIO4 are good choices).

fluffymadness, 2021-05-05:
Ah ok. will stay on on the nrf module then.
But still thanks for the help with wiring.

closed 2021-05-05 by fluffymadness