sidoh/esp8266_milight_hub #723: Home Assistant update change MQTT light settings

poudenes, 2021-04-16:
They announced some changes and this is written:

```
MQTT JSON light now supports color_mode which should be used together 
with supported_color_modes to signal the light’s features. 

Feature flags color_temp, hs, rgb, white_value, xy are all deprecated and support 
will be removed in 2021.10.
```
https://www.home-assistant.io/integrations/light.mqtt/
```
  - platform: mqtt
    schema: json
    name: mqtt_json_light_1
    state_topic: "home/rgb1"
    command_topic: "home/rgb1/set"
    brightness: true
    color_mode: true
    supported_color_modes: ["rgb"]
```

```
color_mode boolean (optional, default: false)
Flag that defines if the light supports color modes.
```

```
supported_color_modes list (optional)
A list of color modes supported by the list. This is required if color_mode is True. Possible color modes are onoff, brightness, color_temp, hs, xy, rgb, rgbw, rgbww.
```

In the settings part in MQTT we have **Group state fields**
Is that the supported_color_modes?

So the wiki must be updated also when we know how the MQTT part will be setup:
https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant

sidoh, 2021-04-16:
Makes sense. Also means updating the MQTT discovery code.

If you have tested with the new syntax, happy to just copy/paste any edits or additions you have.

poudenes, 2021-04-17:
```
- platform: mqtt
  name: "Milight TEST"
  command_topic: "Milight_Hub/0xAAAF/rgb_cct/0"
  state_topic: "Milight_Hub/states/0xAAAF/rgb_cct/0"
  schema: json
  color_mode: true
  supported_color_modes: ["rgbww"]
  brightness: true
  effect: true
  qos: 1
  optimistic: true
  retain: false
  availability_topic: "Milight_Hub/client_status"
  payload_available: "connected"
  payload_not_available: "disconnected"
  effect_list:
    - "night_mode"
    - "white_mode"
    - "0"
    - "1"
    - "2"
    - "3"
    - "4"
    - "5"
    - "6"
    - "7"
    - "8"
```
```
 Exception in state_received when handling msg on 'Milight_Hub/states/0xAAAF/rgb_cct/0': '{"state":"OFF","status":"OFF","brightness":255,"level":100,"kelvin":88,"color_temp":344,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}'
```

in the JSON that comes from the hub there is no **color_mode** 

this is what the page say on HA about JSON:

**JSON schema**

The mqtt light platform with JSON schema lets you control a MQTT-enabled light that can receive JSON messages.

This schema supports on/off, brightness, RGB colors, XY colors, color temperature, transitions, short/long flashing and white values. The messages sent to/from the lights look similar to this, omitting fields when they aren’t needed. The color_mode will not be present in messages sent to the light. It is optional in messages received from the light, but can be used to disambiguate the current mode in the light. In the example below, color_mode is set to rgb and color_temp, color.c, color.w, color.x, color.y, color.h, color.s` will all be ignored:

```
{
  "brightness": 255,
  "color_mode": "rgb",
  "color_temp": 155,
  "color": {
    "r": 255,
    "g": 180,
    "b": 200,
    "c": 100,
    "w": 50,
    "x": 0.406,
    "y": 0.301,
    "h": 344.0,
    "s": 29.412
  },
  "effect": "colorloop",
  "state": "ON",
  "transition": 2,
  "white_value": 150
}

```
What I understand is that there must be a color_mode in the JSON.

poudenes, 2021-09-06:
Hi all,

With HA 2021.9.2 it seems that control the bubs not work correct anymore.

gzxto1, 2021-09-06:
No Problem, with 2021.9.3

poudenes, 2021-09-06:
When I change bulbs from warm white to effect 0 (color changes) the saturation is not fully colored. The warm white leds are still on also.

closed 2023-04-13 by poudenes