sidoh/esp8266_milight_hub #724: Random intercepting FUT089 (but works OK with B8 and FUT087)

lelkluk, 2021-04-23:
Hi everyone. I've got a problem in intercepting my two FUT089. It works about 30% ot the time, in a random way.
On the other hand I don't have any problem with my numerous FUT087 nor with my B8 panel. They are each time correctly intercepted
I have no problem either when transmitting towards the bulbs. My bulbs are correctly controlled by the combo ESP8266/NRF24L01 100% of the time

All my tests are done at very short range, without noticeable disturbance.

I use the nodeMCU V2 (ESP8266-12E) and NFR24L01+ SMD
I tried soldering 2 capacitors 100nF and 4,7 uF just on the NRF24L01 board with no change
I tried several power supply, just in case
I flashed with the file esp8266_milight_hub_esp12-1.10.7.bin

I have absolutely no problem when my two FUT089  directly drive my 18 bulbs, no problem with my strip controller WL5 and with my two WL-Box1. Everything works very fine as expected.

The problem is only with the combination FUT089 -> ESP8266/NRF14L01

To  highlight the problem I just have to observe the LED of the NodeMCU board. No matter my Home assistant, my MQTT etc.
When I press a bouton of either of my FUT089, this LED doesn't react every time and of course the sniffer sniffes nothing when the LED doesn't react (but at the same time the bulb reacts as excepted each time). 
When I use the B8 panel, and all my numerous FUT087, the LED properly reacts each time I press something.

I tried to change the setting of the "Radio" tab  during a full evening with no change in this behavior

What can I do? Is this a known weakness? Do I do something bad?
I hope I have clearly described my problem, as English is not my common language

sidoh, 2021-04-23:
Hi @lelkluk, 

That's certainly strange. An obvious answer does not spring to mind.

I might suggest enabling debug logging and opening a serial console. (uncomment [this line](https://github.com/sidoh/esp8266_milight_hub/blob/master/platformio.ini#L41) and build with platformio)

> I tried to change the setting of the "Radio" tab during a full evening with no change in this behavior

By this I assume you mean you changed the listen channel? This setting?

[image]

lelkluk, 2021-04-23:
Thank you for your answer.
Yes I tried to change this Listen setting, as well as the others, just in case...
I dont know if this is necessary, but I restarted after every change of setting, as I saw no différence.
May be there is a difference for a long range communication, but I am 0,3m to 2m far during my tests.

I could try what you suggest about the debug logging, but I am afraid no being able to do this easily, due to my inexperience. 

I could try to flash again, as I own only one nodeMCU (but I have another NRF24), but to be honest even that I am not sure to be able to do this quickly, I'll have to look for a right tuto...
I will order another combo, that I will receive some day, I am not hurry

But you confirm that what I described is not normal, so I will insist on trying to understand and make progress

Thank you for your job, for sharing, for helping  :)

DOCaCola, 2021-04-24:
I also have issues reading key inputs sent from the FUT89. Milight hub doesn't receive about 1 in 6 messages (simple on/off commands).
I actually bought a second FUT89 (i thought the first one might be faulty) and the issue is exactly the same with both of them. Changing channels did not improve the situation for me. Commands sent directly lights themselves all work without issues with this remote.
I also have a FUT87 and FUT092 that both work very well with no messages dropped.

I am currently using a NRF24L01 with external antenna.

lelkluk, 2021-04-25:
DOCaCola it seems that we are exactly in the same boat. I have ordered from Amazon some extra nodeMCU and NRF24L01 to quickly evaluate others combos, possibly from another provider.

lelkluk, 2021-04-25:
I received a  new nodeMCU V2 from Amazon (this one is annotated ESP8266MOD instead ESP8266-12E the first one) and I had one NFR24L01+ SMD left to test with
I have flashed with the same esp8266_milight_hub_esp12-1.10.7.bin

I have exactly the same issue with this new combo.
Everything seems to work fine except reading correctly the FUT089 keys
In two days I should receive from Amazon some NRF24L01+ (the normal ones, not SMD this time) to do a new test with

lelkluk, 2021-04-25:
Good news! There is no more FUT089 reading keys problem when flashing the ESP with the old file esp8266_milight_hub_esp12-1.10.6.bin

My two FUT089 are perfectly red 100% of the time with this release

DOCaCola, 2021-04-25:
Yes, 1.10.6 doesn't have the issue. I just downgraded and the problem with the FUT089 are entirely gone.

Vincentiano, 2021-08-10:
Thank you guys for the suggestion, this worked wonders. I have been bumming out about this for quite a while and couldn't find an answer. I really like the fut089 remote for the amount of functionality, so I am really happy with this solution.

DOCaCola, 2021-12-02:
Yes. I also compared the code changes a while back and found nothing that may cause the issue with this particular remote. So i was surprised to find that the older version actually worked for whatever reason. I have a collection of all different milight remotes that worked just fine with the later version, but just not the FUT089.

sidoh, 2021-12-02:
Wow, totally missed the notifications for this from August, sorry guys.

I don't see any changes that should cause this sort of issue. If someone with a remote could try building 1.10.6 from source, ensuring issue is still fixed, and then adding changes from 1.10.7 in piecewise that's probably the best way to narrow down what's going on.