sidoh/esp8266_milight_hub #727: Add input for switch or button

Eligio-Greco, 2021-05-12:
Thanks to your fantastic work I can turn on the lights with the Milight remote control, with Home Assistant, with Google, but my mother is used to the classic wall switch: it would be possible to add an input on the ESP (to which to connect a switch or a button) to assign to a device?

Thank you, Eligio Greco.

garmck, 2021-05-12:
You can use these products
https://www.futlight.com/light/products/panel_remote.html
The B# products are battery powered and can be attached to a wall or next to an existing light switch.

sidoh, 2021-05-13:
Hi @Eligio-Greco!

There's currently no support for an onboard button to control lights. This has been requested a handful of times (see #599) in the past, but I've been hesitant about implementing it.

As @garmck suggests, you can use one of the milight products and set up espMH as a passthrough. Some instructions on that here: https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

Since there's already an open issue for this request, gonna close this as a duplicate.

closed 2021-05-13 by sidoh