sidoh/esp8266_milight_hub #729: Issue transmitting and receiving FUT089 / B8 packets

philicibine, 2021-05-19:
### Describe the bug

I have paired my bulbs with my B8 remote.  And also sniffed the traffic below:

```
fut089 packet received (9 bytes):
Raw packet: D9 D9 52 3A A1 6D 1E 9F A0 

Decoded:
Key      : D9
b1       : 25
ID       : 1644
Command  : 01
Argument : 14
Sequence : D9
Group    : 04
Checksum : 12
fut089 packet received (9 bytes):
Raw packet: 63 8C 14 7C 03 42 0F 18 32 

Decoded:
Key      : 63
b1       : 25
ID       : 1644
Command  : 05
Argument : 64
Sequence : D8
Group    : 04
Checksum : DB
fut089 packet received (9 bytes):
Raw packet: 5F 90 18 80 07 20 08 1C 25 

Decoded:
Key      : 5F
b1       : 25
ID       : 1644
Command  : 05
Argument : 5E
Sequence : D7
Group    : 04
Checksum : D0
fut089 packet received (9 bytes):
Raw packet: 72 35 F5 CA 1E 3E 41 DD E0 

Decoded:
Key      : 72
b1       : 25
ID       : 1644
Command  : 05
Argument : 4A
Sequence : D4
Group    : 04
Checksum : AC
fut089 packet received (9 bytes):
Raw packet: CA DD DD 32 06 E1 93 C5 C5 

Decoded:
Key      : CA
b1       : 25
ID       : 1644
Command  : 05
Argument : 1D
Sequence : CE
Group    : 04
Checksum : 61
fut089 packet received (9 bytes):
Raw packet: D0 08 1B 0F 92 0A 2F 91 16 

Decoded:
Key      : D0
b1       : 25
ID       : 1644
Command  : 05
Argument : 16
Sequence : CD
Group    : 04
Checksum : 53
fut089 packet received (9 bytes):
Raw packet: AB 44 EC B4 BB 9A 49 D0 4C 

Decoded:
Key      : AB
b1       : 25
ID       : 1644
Command  : 05
Argument : 04
Sequence : CA
Group    : 04
Checksum : 25
fut089 packet received (9 bytes):
Raw packet: 08 D0 E3 57 5A C7 6B 59 0B 

Decoded:
Key      : 08
b1       : 25
ID       : 1644
Command  : 05
Argument : 03
Sequence : C9
Group    : 04
Checksum : 04
fut089 packet received (9 bytes):
Raw packet: 11 A1 7A 42 A5 85 47 A7 6D 

Decoded:
Key      : 11
b1       : 25
ID       : 1644
Command  : 05
Argument : 04
Sequence : C8
Group    : 04
Checksum : FD
fut089 packet received (9 bytes):
Raw packet: 0C CC DF 53 56 BE 69 55 BC 

Decoded:
Key      : 0C
b1       : 25
ID       : 1644
Command  : 05
Argument : 06
Sequence : C7
Group    : 04
Checksum : F9
fut089 packet received (9 bytes):
Raw packet: 38 A0 93 A7 2A 7E C3 29 FE 

Decoded:
Key      : 38
b1       : 25
ID       : 1644
Command  : 05
Argument : 1A
Sequence : C1
Group    : 04
Checksum : E3
fut089 packet received (9 bytes):
Raw packet: 34 A4 97 AB 2E 7E BE 2D F9 

Decoded:
Key      : 34
b1       : 25
ID       : 1644
Command  : 05
Argument : 1E
Sequence : C0
Group    : 04
Checksum : E2
fut089 packet received (9 bytes):
Raw packet: 43 AC 74 5C 63 22 0C 78 FB 

Decoded:
Key      : 43
b1       : 25
ID       : 1644
Command  : 05
Argument : 24
Sequence : BF
Group    : 04
Checksum : E2
fut089 packet received (9 bytes):
Raw packet: 3E 69 49 BE 92 4C 3F 51 CC 

Decoded:
Key      : 3E
b1       : 25
ID       : 1644
Command  : 05
Argument : 2C
Sequence : BE
Group    : 04
Checksum : E4
fut089 packet received (9 bytes):
Raw packet: 8D 25 FE C6 25 09 6E 2B 54 

Decoded:
Key      : 8D
b1       : 25
ID       : 1644
Command  : 01
Argument : 04
Sequence : B5
Group    : 04
Checksum : 62
fut089 packet received (9 bytes):
Raw packet: 78 60 53 E7 EE 59 8E EE A9 

Decoded:
Key      : 78
b1       : 25
ID       : 1644
Command  : 01
Argument : 01
Sequence : B4
Group    : 01
Checksum : 76
```
I have also tried to spoof this traffic with the hub.  it seems to transmit, but the lights dont react and it seems to slow down and crash.  The traffic it sends is as below:
```
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F E6 D5 E5 61 F7 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 81
Argument : 0D
Sequence : 57
Group    : 04
Checksum : 20
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 66 BC E4 61 71 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 01
Argument : 14
Sequence : 56
Group    : 04
Checksum : A6
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 66 BC E7 61 74 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 01
Argument : 14
Sequence : 55
Group    : 04
Checksum : A5
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 66 CC E6 61 83 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 01
Argument : 04
Sequence : 54
Group    : 04
Checksum : 94
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 60 D3 E9 61 73 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 07
Argument : 0F
Sequence : 53
Group    : 04
Checksum : A4
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 60 CE E8 61 81 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 07
Argument : 02
Sequence : 52
Group    : 04
Checksum : 96
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 60 D0 EB 65 9A 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 07
Argument : 00
Sequence : 51
Group    : 00
Checksum : 8F
fut089 packet received (9 bytes):
Hub traffic

Raw packet: 00 D8 CB 5F 60 D5 EA 65 8E 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 07
Argument : 0D
Sequence : 50
Group    : 00
Checksum : 9B
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 60 D3 FD 65 8B 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 07
Argument : 0F
Sequence : 4F
Group    : 00
Checksum : 9C
fut089 packet received (9 bytes):
Raw packet: 00 D8 CB 5F 60 D1 FC 65 9C 

Decoded:
Key      : 00
b1       : 25
ID       : 1644
Command  : 07
Argument : 01
Sequence : 4E
Group    : 00
Checksum : 8D
```

#### Firmware version
most recent and 1.10.6 tried so far

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.0.12","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":17088,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":180,"http_repeat_factor":5,"auto_restart_period":0,"mqtt_server":"192.168.0.16:1883","mqtt_username":"pi","mqtt_password":"xxxxx","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":30,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"HIGH","wifi_static_ip":"192.168.0.12","wifi_static_ip_gateway":"192.168.0.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","HIGH"],"device_ids":[5700,36740,39691,39692,39693],"gateway_configs":[[39691,3389,6],[36740,3390,6],[5700,3391,5]],"group_state_fields":["state","level","hue","saturation","mode","color_temp","bulb_mode"],"group_id_aliases":{"Office Light":["rgb_cct",39693,1]}}
```
If you need any extra output please let me know.  Hope to get this sorted. Thanks

Linkenelis, 2021-05-19:
"I have paired my bulbs with my B8 remote."
Did you pair it with the MiLight hub also?

philicibine, 2021-05-19:
I have just tried this and the bulbs don't respond.  They don't pair at all with the hub

Linkenelis, 2021-05-19:
I am working on Milighthub on esp32 with display and battery, so not the original setup...
My wife does not want me to touch the original :-)
Relevant MiLight code is the same though.
So I tried the FUT89 / B8 and it works a expected via webpage. From the display it uses REST API to command the MiLight hub and also works OK. See my settings ,"testlamp":["fut089",3,3]}} (device id=3 group =3)

I noticed from your settings:

- "group_id_aliases":{"Office Light":["rgb_cct",39693,1]}}
There is only 1 device registered "Office Light" on device ID 39693 with Group ID 1 and remote type = "rgb-cct". (So only 1 lamp **can** be paired and it is not the FUT89 / B8...
- "device_ids":[5700,36740,39691,39692,39693]; 5 device id's, of which 3 have gateways... "gateway_configs":[[39691,3389,6],[36740,3390,6],[5700,3391,5]]; 2 type 6 and 1 type 5

I sugest you go to the web interface and first remove:

- Office Light
- All device id's
- All the UDP on settings (these are sometimes needed for 3rd party like home assistant via plugin)
- Start over

Create a Device Id, select a remote type, select a group nbr and create a name for the device (=lamp).
As soon as you give it a name in Device Name, it will be stored.
Now you can pair and try again.

my settings: {"admin_username":"","admin_password":"","ce_pin":26,"csn_pin":5,"reset_pin":0,"led_pin":27,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Fast blip","led_mode_packet_count":4,"hostname":"milight-hub-esp32","rf24_power_level":"MAX","rf24_listen_channel":"MID","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["MID"],"device_ids":[3,10],"gateway_configs":[[3,5986,6]],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"test 1 3":["cct",3,1],"test3 10":["rgb_cct",10,3],"test4 10":["rgb_cct",10,4],"testlamp":["fut089",3,3]}}

philicibine, 2021-05-19:
Hi, so i have done all the above.  I have basically removed everything and started fresh.  I added the device ID of the remote and gave it a name.  Saved it and then tried to re-pair.  The light still doesn't respond. As i said before it used to when using a device ID from a standard remote.

philicibine, 2021-05-19:
To add to this.... i compiled the source with -D DEBUG PRINTF line uncommented and hooked it up to my pc.

It seems like any button press on the webpage results in hundreds of packets being sent and ultimately slowing the hub down. 

Here is the serial log for one button press
```
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 25
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 24
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B6 61 53 
Elapsed: 23
[... 933 lines cut ...]
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 22
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 24
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 23
Sending packet (10 repeats): 
00 D8 CB 5F 66 CC B5 61 52 
Elapsed: 23
```

Linkenelis, 2021-05-19:
"Saved it and then **tried** to re-pair". It did not pair, I guess?
 "I added the device ID of the remote". This can be any number (well nor really high ones), but 1, 2 or 3 is OK.
Just to make sure it is not the remote type...
Can you select a different group nbr; select RGB-CCT as a remote type; add another name and try to pair that one?
The RGB-CCT is the same as the FUT89/B8 but for a group of 4. I think this will not pair also.
You see what the D1 mini is trying to send, not if it is going through the RF24

I used the "start sniffing" option here.
This is my normal on and off:
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 22 66 D4 0F 64 7C 

Decoded:
Key      : 00
b1       : 25
ID       : 0003
Command  : 01
Argument : 0C
Sequence : BD
Group    : 03
Checksum : AD
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 22 66 CF 0E 64 76 

Decoded:
Key      : 00
b1       : 25
ID       : 0003
Command  : 01
Argument : 03
Sequence : BC
Group    : 03
Checksum : A3

Disconected CE, same on and off:
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 22 66 D4 0D 64 7A 

Decoded:
Key      : 00
b1       : 25
ID       : 0003
Command  : 01
Argument : 0C
Sequence : BF
Group    : 03
Checksum : AF
fut089 packet received (9 bytes):
Raw packet: 00 D8 E1 22 66 CF 0C 64 74 

Decoded:
Key      : 00
b1       : 25
ID       : 0003
Command  : 01
Argument : 03
Sequence : BE
Group    : 03
Checksum : A5

for one buttonclick it gives me 5 messages
21:36:31.204 --> disableCore0WDT()
21:36:33.132 --> Unhandled websocket event: 10
21:36:54.715 --> Enqueuing packet
21:36:54.715 --> Switching to next packet, 1 packets in queue
21:36:54.715 --> Sending packet (10 repeats): 
21:36:54.715 --> 00 DB E1 29 66 D1 86 66 10 
21:36:55.661 --> Elapsed: 960
21:36:55.661 --> Sending packet (10 repeats): 
21:36:55.661 --> 00 DB E1 29 66 D1 86 66 10 
21:36:56.625 --> Elapsed: 960
21:36:56.625 --> Sending packet (10 repeats): 
21:36:56.625 --> 00 DB E1 29 66 D1 86 66 10 
21:36:57.584 --> Elapsed: 960
21:36:57.584 --> Sending packet (10 repeats): 
21:36:57.584 --> 00 DB E1 29 66 D1 86 66 10 
21:36:58.557 --> Elapsed: 960
21:36:58.557 --> Sending packet (10 repeats): 
21:36:58.557 --> 00 DB E1 29 66 D1 86 66 10 
21:36:59.502 --> Elapsed: 960
21:37:26.183 --> get time
21:37:26.183 --> Wednesday, May 19 2021 21:37:26

When I connect CE again and reload the page one buttonclick "On" (notice the difference in elapsed):
21:40:27.914 --> Enqueuing packet
21:40:27.914 --> Switching to next packet, 1 packets in queue
21:40:27.914 --> Sending packet (10 repeats): 
21:40:27.914 --> 00 D8 E1 22 66 CF 1C 64 84 
21:40:27.914 --> Elapsed: 7
21:40:27.914 --> Sending packet (10 repeats): 
21:40:27.914 --> 00 D8 E1 22 66 CF 1C 64 84 
21:40:27.914 --> Elapsed: 7
21:40:27.914 --> Sending packet (10 repeats): 
21:40:27.914 --> 00 D8 E1 22 66 CF 1C 64 84 
21:40:27.914 --> Elapsed: 7
21:40:27.914 --> Sending packet (10 repeats): 
21:40:27.914 --> 00 D8 E1 22 66 CF 1C 64 84 
21:40:27.914 --> Elapsed: 7
21:40:27.914 --> Sending packet (10 repeats): 
21:40:27.914 --> 00 D8 E1 22 66 CF 1C 64 84 
21:40:27.914 --> Elapsed: 7

philicibine, 2021-05-19:
> "Saved it and then **tried** to re-pair". It did not pair, I guess?  
Correct!

> "I added the device ID of the remote". This can be any number (well nor really high ones), but 1, 2 or 3 is OK.
I know this, but I changed to this wall panel remote so that I can use it aswell as my automation software.  So I need the lights to pair with the device ID of the remote.

> Just to make sure it is not the remote type...
> Can you select a different group nbr; select RGB-CCT as a remote type; add another name and try to pair that one?
> The RGB-CCT is the same as the FUT89/B8 but for a group of 4. I think this will not pair also.
Will try this shortly and report back.

> You see what the D1 mini is trying to send, not if it is going through the RF24

So it does look like mine is sending far too many packets.  I did say it seems unresponsive when this happens.  

Will report back shortly to say if they pair with the hub in rgb_cct mode.

philicibine, 2021-05-20:
> > Just to make sure it is not the remote type...
> > Can you select a different group nbr; select RGB-CCT as a remote type; add another name and try to pair that one?
> > The RGB-CCT is the same as the FUT89/B8 but for a group of 4. I think this will not pair also.
> 
> Will try this shortly and report back.

I should have said.. it must be the remote type because these lights have worked before, with this hub.  I have used them for several years with this hub.  Same bulbs.  All i have done is bought this B8 wall panel.

Its something in the software for this remote type.

Linkenelis, 2021-05-20:
I know this, but I changed to this wall panel remote so that I can use it aswell as my automation software. So I need the lights to pair with the device ID of the remote.
?? Could this be the problem ??  I would say all nbrs are OK but not that one. How is your lamp going to see the difference between the 2 (remote and Milight-hub and any other controling device need a different id!)?
So if it is already paired to that id, it will not pair again!!
You can pair with more than 1 device and they (should) have different ID's.
I have a T4 wallswitch, a remote and a MiLight-hub. They are need to be paired individually. They all have different id's (but I never checked the wallswitch nor the remote).

- The original MiLight-hub has 1 device-id; this one can emulate several original ones (more device-id's). So keep it simple call it "1" (it will return a value of 0x1), 2 and 3 (I did not run in to the max of the ESP8266, but I am certain it can represent 3 original ones). For compatibility with home automation systems I use RGB-CCT. 3 x 4 groups of lamps: device id = 1; group = 1 can pair several lamps. That is 12 individual groups to control a number > 12 of lamps.
- Remote type is needed (almost) only for displaying the options; You can control a CCT-lamp with the FUT89/B8 remote (saturation will influence brightness but on/off, pairing and brightness will work)
- The only difference between FUT89/B8 and RGB-CCT is 8 and 4 group size.

philicibine, 2021-05-20:
> You can pair with more than 1 device and they (should) have different ID's.

This was new to me! Thanks for enlightening me!

So now, I have the bulbs paired with the old device ID`s and now also the wall panel remote.  Problem solved for me.  

Has done nothing to explain the issue with the hub tho.  Hopefully someone else will see similar issues and be able to help.

Linkenelis, 2021-05-20:
Hah, glad to help out.
Like said you do not need the old device id's. Just a low number, as the remotes and such normally use higher numbers. So use 1 to 9 for the device id's. 
If you still do not have device id = 1, then create it, use the FUT89/B8 as remote type, select group 8 and call it "test" in device name. Then pair...
Yes with a different device id, you can pair again with the MiLight-hub. Your lamp will think it is a different device.
That will show that it is possible to do so. Just in case you want both systems with the same apearance.