sidoh/esp8266_milight_hub #730: Web interface stops responding

AaronBDO, 2021-05-24:
### Describe the bug

I am having to restart my nodemcu board every 24 hours or so due to the web interface becoming unresponsive.    It also stops responding to ping requests, although it does seem to continue sending / receiving MQTT to home assistant as I can continue to control my bulbs from there.  Given the web interface is not essential once it's configured, this is not a huge issue, but I'm troubleshooting another issue as posted on the HA forums so it's rather annoying.

### Steps to reproduce

I'm using a LOL1n brand NodeMCU board, with NRF24L01.   This seems to be consistent behaviour over any  24 hour period.  I haven't actually timed the exact period it takes to become unresponsive.

### Expected behavior

Web interface should remain available, and pingable

### Setup information

#### Firmware version
1.10.7
issue did exist on  the previous version I was running but I'm sorry I didn't record what those were, I updated to the latest to see if it would resolve.

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.7","ip_address":"192.168.60.8","reset_reason":"External System","variant":"d1_mini_PRO","free_heap":17208,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":100,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.60.2:1883","mqtt_username":"redacted","mqtt_password":"redacted","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":500,"packet_repeat_throttle_threshold":750,"packet_repeat_minimum":100,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"Flicker","led_mode_operating":"On","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":100,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[1,60724],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Hall":["rgbw",1,2],"Master1":["rgbw",60724,1],"Master2":["rgbw",60724,2],"Master3":["rgbw",60724,3],"Misha Bedroom":["rgbw",1,3],"Study":["rgbw",1,1]}}
```

### Additional context
This is my first attempt at configuration of a chip like this, so please excuse me if I don't know what else to give you to assist with diagnosis.

Linkenelis, 2021-05-25:
Not seen this behaviour before. But just maybe your device is busy sending to bulbs and MQTT.
So this might not be a solution, but it is easy to do...
Maybe it makes a difference.

"rf24_channels":["LOW","MID","HIGH"]
"rf24_listen_channel":"LOW"
"packet_repeats_per_loop":100
"packet_repeats":100

maybe your board is "busy"
Sending on 3 channels, is that OK, or will just MID do?
packet_repeats_per_loop: 100 ; set it to 10
packet_repeats:100; set it to 50

I know calculation is not all in 1 go, but just for reference
you are sending a lot of packages 100*100*3channels = 30.000

standard setting 50*10*1 = 500

Linkenelis, 2021-05-25:
hmm looks funny
100x100x3=30.000
50x10x1=500

sidoh, 2021-05-25:
I've not heard this specific set of symptoms before, so sort of taking a stab in the dark, but...

How are the ESP/nRF wired? If you're using jumper cables, make sure the connections aren't loose at all. Consider switching to soldered connections if you can. Sometimes if the connection is shoddy the nRF library will hang stuff.

AaronBDO, 2021-05-25:
Thank you all.

Sure enough after writing this it froze once yesterday, but it's been stable ever since.   Of course I did move it around when I reset it so perhaps it is related to the connections as you suggest - they are just jumpered.     I can try soldering if it starts playing up again.

The packet settings are thus as I was reading someone on another blog found them as the most reliable, and as I was working through another issue as per [HA forum](https://community.home-assistant.io/t/esp8266-milight-hub-activate-nightmode-from-bulb-off-state/310252/8) I thought I'd give it a shot.  I will probably raise that issue here once I wrap my head around mqtt strings and can be a bit more descriptive with my description ;)

Happy to close this off as it's intermittent and if you haven't heard of it it would be something I've done which I appreciate is nigh on impossible to resolve from your pov.

closed 2021-05-25 by AaronBDO