sidoh/esp8266_milight_hub #735: Strange error in HA

poudenes, 2021-08-10:
Hi All,

For long time all my lights working perfect. But at a certain moment i see those errors pop up in the error log.
I dont know how to solve this. Tried many things. Maybe HA has changes some things i dont know.

(it is 1 line but its always more then 1. But show one)
```
2021-08-10 06:23:54 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'Milight_Hub/states/0xAAAB/rgb_cct/1': '{"status":"ON","brightness":"255"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 34, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 299, in state_received
    if values["state"] == "ON":
KeyError: 'state'
```
```

- name: "Bedroom 5"
  command_topic: "Milight_Hub/0xAAAB/rgb_cct/1"
  state_topic: "Milight_Hub/states/0xAAAB/rgb_cct/1"
    platform: mqtt
    schema: json
    color_temp: true
    rgb: true
    brightness: true
    effect: true
    qos: 1
    optimistic: true
    retain: false
    availability_topic: "Milight_Hub/client_status"
    payload_available: "connected"
    payload_not_available: "disconnected"
    effect_list:
      - "night_mode"
      - "white_mode"
      - "0"
      - "1"
      - "2"
      - "3"
      - "4"
      - "5"
      - "6"
      - "7"
      - "8"
```

poudenes, 2021-08-10:
'{"status":"ON","brightness":"255"}'

status is ON and on. Can that be the problem?

garmck, 2021-08-10:
There was a breaking change for MQTT Light integration in April.
[https://www.home-assistant.io/blog/2021/04/07/release-20214/#breaking-changes](url)

[https://github.com/home-assistant/core/pull/47993](url)
@sidoh some features are being deprecated in 2021.10 - will these impact the MQTT JSON currently being used?
Looks like **color_temp** will no longer be used.

poudenes, 2021-08-11:
https://github.com/sidoh/esp8266_milight_hub/issues/723

Posted already things that will change....

poudenes, 2021-08-11:
When this part is in the HUB I get many many errors:

mqtt_topic_pattern - set this to Milight_Hub/:device_id/:device_type/:group_id

poudenes, 2022-10-24:
Any news about this? :)

closed 2023-04-13 by poudenes