sidoh/esp8266_milight_hub #737: Lights pair to remote id by themself

ReinierGielen, 2021-10-05:
### Describe the bug

Hello all, first of all this is a bug but may be not with the hub.
I have had a set up with milights abt the milight)hub for years working perfectly.
I have a few groeps of FUT066 downligts in the livingroom and a few groeps of spots FUT104 in the kitchen.
The lights in the living room are mostly turned on and off by hand and the lights in the kitchen are on a motion sensor.

After an update to the latest version of the hub the livingroom started to respond to the automation of the kitchen.
Returning to the previous version did not help.

I tried repairing all the lights and then every thing works again for a few hours to a few days. Then the living room is responding to the kitchen automation again. I did this several times and also changes the remote id's. It did not help. The livingroom starts responding to one of the lights in the kitchen ( NOT the same light at every attempt ).

The livingroom also still responds to its own remote id's
I did some sniffing, and according to the sniffing the signals send by the automation are from the kitchen and the kitchen only.

Somehow the livingroom lights get paired with one of the kitchen lights by them self.

### Steps to reproduce

No Idea how to reprocude other than to do a normal setup and wait a couple off day's

### Expected behavior

I expect each light to listen only to the remote id I paired it with

### Setup information

#### Firmware version
1.10.7 now ( 1.10.2 before )

### Additional context

Does anyone have a suggestion what might be wrong ?

ReinierGielen, 2021-10-14:
That is exactly how I have set it up.

And when you turn on light.kitchen_counter all lights go on BUT it has NO effect on light.kitchen in the UI of HA.

They respond all as individual lights even the group 0

garmck, 2021-10-14:
Ah, i think I understand now, so you want the status within HomeAssistant to reflect Group 0 use?
So e.g. Group 0 on, 1,2,3,4 should be on. Group 0 off, 1,2,3,4 should be off?