sidoh/esp8266_milight_hub #745: Nothing captured from 2 remotes

Arakon, 2021-11-11:
### What is the model number of the device you're trying to control?
[image]

### What firmware version(s) have you tried? 
Lastest as of today.

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
Wemos D1 Mini.

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGB+CCT I believe.

### Have you tried controlling the device with a physical remote?
Yes. Trying to get the remote recognized by the hub.

[image]

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
<html><body>
<!--StartFragment--><div id="content"><div class="tabs "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><input class="searchBox devtools-filterinput" placeholder="Filter JSON"></div><div class="panelContent">

  |  
-- | --
firmware | "milight-hub"
version | "1.10.8"
ip_address | "192.168.0.89"
reset_reason | "External System"
variant | "d1_mini"
free_heap | 18912
arduino_version | "2_4_2"
queue_stats |  
length | 0
dropped_packets | 0

</div></div></div></div><!--EndFragment-->
</body>
</html>
```

#### /settings

```json
<html><body>
<!--StartFragment--><div id="content"><div class="tabs "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><input class="searchBox devtools-filterinput" placeholder="Filter JSON"></div><div class="panelContent">

  |  
-- | --
admin_username | ""
admin_password | ""
ce_pin | 4
csn_pin | 15
reset_pin | 0
led_pin | -2
radio_interface_type | "nRF24"
packet_repeats | 50
http_repeat_factor | 1
auto_restart_period | 0
mqtt_server | ""
mqtt_username | ""
mqtt_password | ""
mqtt_topic_pattern | ""
mqtt_update_topic_pattern | ""
mqtt_state_topic_pattern | ""
mqtt_client_status_topic | ""
simple_mqtt_client_status | false
discovery_port | 48899
listen_repeats | 3
state_flush_interval | 10000
mqtt_state_rate_limit | 500
mqtt_debounce_delay | 500
mqtt_retain | true
packet_repeat_throttle_sensitivity | 0
packet_repeat_throttle_threshold | 200
packet_repeat_minimum | 3
enable_automatic_mode_switching | false
led_mode_wifi_config | "Fast toggle"
led_mode_wifi_failed | "On"
led_mode_operating | "Slow blip"
led_mode_packet | "Flicker"
led_mode_packet_count | 3
hostname | "milight-hub"
rf24_power_level | "MAX"
rf24_listen_channel | "HIGH"
wifi_static_ip | ""
wifi_static_ip_gateway | ""
wifi_static_ip_netmask | ""
packet_repeats_per_loop | 10
home_assistant_discovery_prefix | ""
wifi_mode | "n"
default_transition_period | 500
rf24_channels |  
0 | "LOW"
1 | "MID"
device_ids | []
gateway_configs | []
group_state_fields |  
0 | "state"
1 | "brightness"
2 | "mode"
3 | "color_temp"
4 | "bulb_mode"
5 | "computed_color"
group_id_aliases | {}

</div></div></div></div><!--EndFragment-->
</body>
</html>
```

There's absolutely no reaction while sniffing, I tried 2 different (known working) remotes. The remotes seem the same or similar to the ones I've seen used for this project, so I'd expect to see at least some kind of data flow.
Any ideas here? I also tried 2 different NRF24L01 modules.

sp5mbx, 2021-11-11:
I was helped by adding 220 microF / 10V capacitors on the power supply, soft: milight_hub_d1_mini-1.10.8.bin
[image]
73!

Arakon, 2021-11-12:
Thanks, but that didn't help for me, and I tried multiple (known clean) power supplies, too.

sidoh, 2021-11-14:
If you're sure the ESP8266/nRF24 setup is working (i.e., it functions as you expect with other remotes), it's likely that the remote you're using is a knockoff with slightly different RF parameters. It's unfortunately pretty impossible for me to help with this since I can't reproduce locally.

DOCaCola, 2021-12-02:
I could be wrong, but anything designated with that "2.4G" label uses a different protocol than the usual milight remotes

chuckythecat, 2023-11-28:
I have the _**exact**_ same remote and ceiling light, with identical internals and everything, and I also have a RGB strip in my living room, with remote that looks like FUT098. I can capture packets from the FUT098 remote and control the RGB strip in my living room just fine (so looks like my ESP8266+NRF setup is working ok), but I can't capture packets from this FUT096 lookalike ceiling light remote, or control the ceiling light itself from ESP8266, unfortunately.
[image]
[image]
[image]
So is it a knockoff remote after all? Could it be that the protocol itself is pretty much identical, and if one can tweak some settings this remote and ceiling light could work with milight hub? Or is it a different protocol entirely?