sidoh/esp8266_milight_hub #746: HA Discovery not working

poudenes, 2021-11-11:
### Describe the bug

I add "homeassistant" to the config to let auto discovery my lights. But when I check with MQTT Explorer the lights not become into the /homeassistant/light/... etc. Even reboot the hub didnt send the lights so they become discoverable.
 
### Steps to reproduce

Run long time the MQTT config without "homeassistant" discovery filled in. But I thought when add this the light will become discoverable and added to HA.

### Expected behavior

After adding discovery prefix to the MQTT part config directly I see all lights under /homeassistant/...

### Setup information

#### Firmware version
1.10.8 (nodemcuv2)
HAOS: 6.6
Supervisor: -2021.10.8
Core: 2021.11.2

#### Output of http://milight-hub.local/about

```
{"firmware":"milight-hub","version":"1.10.8","ip_address":"192.168.100.70","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":16016,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":75,"http_repeat_factor":5,"auto_restart_period":0,"mqtt_server":"192.168.100.23:1883","mqtt_username":"MYUSER","mqtt_password":"MYPASSWORD","mqtt_topic_pattern":"Milight_Hub/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"Milight_Hub/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"Milight_Hub/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":0,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":20,"packet_repeat_throttle_threshold":750,"packet_repeat_minimum":100,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"On","led_mode_wifi_failed":"Slow toggle","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.68.100.70","wifi_static_ip_gateway":"192.168.100.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":75,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"g","default_transition_period":250,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707,43708],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type","hex_color"],"group_id_aliases":{}}
```

### Additional context

Documentation of HA about MQTT Discovery:
https://www.home-assistant.io/docs/mqtt/discovery/

poudenes, 2021-11-11:
Ok. Found out that I MUST add a device name to let discovery work. When I add a name the discovery was added.

Will it be possible to have this without device name? So device_id will be used?

poudenes, 2021-11-13:
Did a test to use the discovery part manual and its working when I publish this to MQTT Broker.
I used the device_id+groupnumer as name.

If you can implement this when there is no device name given then it will use device_id+groupnumber for example?

mosquitto_pub -h 192.168.100.23 -u hassiomqtt -P abercrombie -p 1883 -t "homeassistant/sensor/vpn_ip_checker/config" -m '

{
"schema": "json",
"name": "0xAAA1_0",
"command_topic": "Milight_Hub/0xAAA1/rgb_cct/0",
"state_topic": "Milight_Hub/states/0xAAA1/rgb_cct/0",
"uniq_id": "7974FE-a1",
"device": {
"manufacturer": "esp8266_milight_hub",
"sw_version": "1.10.8",
"identifiers": [
7959806,
43681,
"rgb_cct",
0
]
},
"availability_topic": "Milight_Hub/client_status",
"payload_available": "connected",
"payload_not_available": "disconnected",
"brightness": true,
"effect": true,
"effect_list": [
"night_mode",
"white_mode",
"0",
"1",
"2",
"3",
"4",
"5",
"6",
"7",
"8"
],
"rgb": true,
"color_temp": true,
"max_mireds": 370,
"min_mireds": 153
}'

sidoh, 2021-11-14:
I think the HASS integration only works with devices you've given an alias. Can you not just give the device an alias?

poudenes, 2021-11-14:
When I do that the memory get full and I lost the config

As you can see I did a manual discovery with a given name. This worked.

So if you can add the device_id+group number then you're done.

The rest is the same as normal way you do when you have a device_name.

We can try it with a beta version?

sidoh, 2021-11-14:
This is a bigger change than it sounds like. Personally think the effort would better be spent splitting the device alias management out of settings. You're welcome to take a crack at it, though!

poudenes, 2021-11-14:
I don't have knowledge about coding, so I can't help you with that... Im only a happy user and try to help!

poudenes, 2021-11-16:
I tried to add to every group a short device name

The first 26 groups went ok.  The last 11 lost the device name and there will be no discovery publish happen also.
Now I can try to make the device name more shorter to add only 1, 2, 3, 4 or 0 to it. But then you always need to change the name in HA to know what number is what bulb in the house...

`{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":75,"http_repeat_factor":5,"auto_restart_period":0,"mqtt_server":"192.168.100.23:1883","mqtt_username":"myusername","mqtt_password":"mypassword","mqtt_topic_pattern":"Milight_Hub/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"Milight_Hub/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"Milight_Hub/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":0,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":20,"packet_repeat_throttle_threshold":750,"packet_repeat_minimum":100,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"On","led_mode_wifi_failed":"Slow toggle","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.68.100.70","wifi_static_ip_gateway":"192.168.100.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":75,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"g","default_transition_period":250,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707,43708],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type","hex_color"],"group_id_aliases":{"a1_1":["rgb_cct",43681,1],"a1_2":["rgb_cct",43681,2],"a1_3":["rgb_cct",43681,3],"a1_a":["rgb_cct",43681,0],"a2_1":["rgb_cct",43682,1],"a2_2":["rgb_cct",43682,2],"a2_3":["rgb_cct",43682,3],"a2_a":["rgb_cct",43682,0],"a3_1":["rgb_cct",43683,1],"a4_1":["rgb_cct",43684,1],"aa_1":["rgb_cct",43690,1],"aa_2":["rgb_cct",43690,2],"aa_3":["rgb_cct",43690,3],"aa_4":["rgb_cct",43690,4],"ab_1":["rgb_cct",43691,1],"ab_2":["rgb_cct",43691,2],"ac_1":["rgb_cct",43692,1],"ac_2":["rgb_cct",43692,2],"ac_3":["rgb_cct",43692,3],"ac_4":["rgb_cct",43692,4],"ac_a":["rgb_cct",43692,0],"ad_1":["rgb_cct",43693,1],"ad_2":["rgb_cct",43693,2],"ad_3":["rgb_cct",43693,3],"ad_4":["rgb_cct",43693,4],"ad_a":["rgb_cct",43693,0]}}`

poudenes, 2021-11-16:
Created a full list of all my 37 bulbs in a file and execute them from command line:
example of 1 line of the 37:
```
# HALLWAY
mosquitto_pub -h 192.168.100.23 -u hassiomqtt -P abercrombie -p 1883 -t "homeassistant/light/milight_hub/0xAAA3_1/config" -m '
{"schema": "json",
"name": "Milight Hallway All",
"command_topic": "Milight_Hub/0xAAA3/rgb_cct/1",
"state_topic": "Milight_Hub/states/0xAAA3/rgb_cct/1",
"uniq_id": "44589e60-f210-405a-827a-26b0c088221a",
"device": {"manufacturer": "esp8266_milight_hub","sw_version": "1.10.8","identifiers": [7959806,43683,"rgb_cct",0]},
"availability_topic": "Milight_Hub/client_status",
"payload_available": "connected",
"payload_not_available": "disconnected",
"brightness": true,
"effect": true,
"effect_list": ["night_mode","white_mode","0","1","2","3","4","5","6","7","8"],
"rgb": true,
"color_temp": true,"qos": 1,"optimistic": true,"retain": false,"qos": 1,"optimistic": true,"retain": true,
"max_mireds": 370,
"min_mireds": 153
}'
```
All bulbs are discovered by HA and working perfect.

- Manual given a name
- unique_id generated in VSC

So it doesn't matter if you use a other value then device_name.

poudenes, 2022-10-24:
Does someone have the knowledge to  check this and see how we can make this better? I'm not a programmer...

bluebier, 2022-10-31:
Hello!
I wrote a node red flow.
Which adds all entities from an object.

`[
    {
        "id": "3ac62bc801a24bbc",
        "type": "inject",
        "z": "84957e642a472b6d",
        "name": "",
        "props": [],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": "1",
        "topic": "",
        "x": 230,
        "y": 260,
        "wires": [
            [
                "b53c8c3f549ed3d6"
            ]
        ]
    },
    {
        "id": "931905c4ff37dacb",
        "type": "counter-loop",
        "z": "84957e642a472b6d",
        "name": "",
        "counter": "i",
        "counterType": "msg",
        "reset": true,
        "resetValue": "value-null",
        "initial": "0",
        "initialType": "num",
        "operator": "lte",
        "termination": "counter",
        "terminationType": "msg",
        "increment": 1,
        "incrementType": "num",
        "x": 420,
        "y": 80,
        "wires": [
            [],
            [
                "df4572928e974e6b"
            ]
        ]
    },
    {
        "id": "df4572928e974e6b",
        "type": "function",
        "z": "84957e642a472b6d",
        "name": "számolás",
        "func": "msg.controller = msg.controllers.controllers[msg.i];\n\nmsg.counter_sub = Object.keys(msg.controller.devices).length - 1;\n\n\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 660,
        "y": 80,
        "wires": [
            [
                "308c18e11312658c"
            ]
        ]
    },
    {
        "id": "ab2b20dead3debf4",
[... 162 lines cut ...]
        "name": "local",
        "broker": "192.168.39.2",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]`

closed 2023-04-13 by poudenes