sidoh/esp8266_milight_hub #747: Saturation control with 5050 RGBW RGB+Warm White

emailej, 2021-11-13:
Apologies if this is naive question but new to the topic and still learning. This is a general question but possibly a feature request around ability to control saturation on 5050 RGBW RGB+Warm White.

I'm using An ESP8266 (NodeMCU) with NRF24L01+ with setup followed exactly from https://github.com/sidoh/esp8266_milight_hub and a FUT038 RGBW controller. Happy to provide more details if needed.

This generally works flawlessly and integrated perfectly with HomeAssistant via MQTT. The only feature missing seems to be ability to control saturation. Is this a limitation of my FUT038 controller? Previously was using a Miboxer WL5,  which doesn't work with HomeAssistant, but does support saturation control.

So is there any way to get saturation control working with this amazing esp8266_milight_hub? That feature was significant in the design of my project.

Thanks!

emailej, 2021-11-14:
Actually the Miboxer WL5 DOES pair with this gateway and DOES work with HomeAssistant and I now have saturation working using FUT089 / B8 protocol . . . although the behavior is a bit odd but functional enough for me :)

sidoh, 2021-11-14:
What remote type did you use when pairing with the controller? You'd need to use RGB_CCT to be able to control saturation.

emailej, 2021-11-15:
Thanks!
That was my mistake. I originally paired the FUT038 with RGBW remote type. After re-pairing with RGB_CCT saturation control is working. As noted above saturation control also works with the Miboxer WL5 paired with FUT089 /B8 remote type. I did not try pairing the WL5 with RGB_CCT so that may work as well.

Hardware I used is shown below in case that makes life easier for anyone. Can be a bit confusing when you're first trying to piece things together.

Cheers!
https://www.amazon.com/dp/B082VLTTTX?psc=1&ref=ppx_yo2_dt_b_product_details
https://www.amazon.com/dp/B010N1SPRK?psc=1&ref=ppx_yo2_dt_b_product_details
https://www.amazon.com/dp/B07CG4HM23?psc=1&ref=ppx_yo2_dt_b_product_details
https://www.ebay.com/itm/194227778258

[image]
[image]