sidoh/esp8266_milight_hub #750: Reboot/Disconnect in loop after adding bulbs

MrSiO, 2021-11-24:
Hello.

I'm having an issues with my hub that began recently after adding 5 rgb_cct to my setup that already contained 3 rgbw bulbs.

Originally I was running on 1.10.7 (d1_mini_PRO) but upgraded to 1.1.8 around the same time I added my 5 new bulbs.

(Downgraded back to 1.10.7 but no luck - same issue)

**Symptoms** : the hub is now rebooting/loosing connectivity in loop every 1 to 3 minutes.

I get the following error trough MQTT:

```

{"status":"disconnected_unclean","firmware":"milight-hub","version":"1.10.7","ip_address":"10.10.107.14","reset_reason":"Software/System restart"}
```

After a hard reset (USB unplugged and replug) I now have this slightly different message:

```
{"status":"disconnected_unclean","firmware":"milight-hub","version":"1.10.7","ip_address":"10.10.107.14","reset_reason":"External System"}
```

I can see from my wifi ap that the connection uptime never gets over 3 minutes.

_Q: I would not mind wiping the config and starting over, but my bulbs are up high on the roof and don't want to have to climb up there to pair again. If I factory reset and then set back the same 0x1234 ID, would I still need to pair back?_ 

Thx alot.

MrSiO, 2021-11-24:
/about

> {"firmware":"milight-hub","version":"1.10.7","ip_address":"10.10.107.14","reset_reason":"External System","variant":"d1_mini_PRO","free_heap":16072,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}

sidoh, 2021-11-24:
Are you able to get serial logs? Hard to tell what's going on without those.

Can confirm you won't need to re-pair lights if you wipe the ESP.

MrSiO, 2021-11-24:
I will start with a factory reset and get the serial infos if not fixed.

Fun fact: I just noticed that the Hub is only having issues when my lights bulbs are ON?

All day hub was up. 6h uptime.
Night comes, automation turns on all my miLights: Hub drops.

I'll Digg into this.

MrSiO, 2021-11-25:
FYI I'm closing this issue.
It turns out many led bulbs and smart bulbs are notorious for interference in the 2.4Ghz range.

This is clearly my issue (just installed those for Christmas...)

Not sure what my next move will be.
Thx for your time.

closed 2021-11-25 by MrSiO

sidoh, 2021-11-25:
Oh yikes, that's no good! For what it's worth, I've not had this problem with any of my bulbs.

Glad you've got it figured out.

MrSiO, 2021-11-25:
Ya.
And you know what is even weirder?
When the bulbs are in white mode or fixed color mode: no problem.

But when they are set to mode (my case I was set to mode 0 - disco) then
the interference is making my 2.4ghz drop.

For the records: I'm using all Unifi product. AP lite for wifi.

Go figure .

sidoh, 2021-11-25:
Oh, interesting. 

The only time Milight bulbs transmit anything is during animation effects, presumably to synchronize transitions with other bulbs.

MrSiO, 2021-11-25:
Ho really?
I was looking for documentation on that.

That's probably my issue

Thx again for your help.

If I find a good workaround I'll let you know.

sidoh, 2021-11-25:
Haha yeah I only know about it because I noticed it show up in radio logs while I was working on reverse-engineering the protocol. Have some notes in an [old issue](https://github.com/sidoh/esp8266_milight_hub/issues/83#issuecomment-312920521). Unfortunately not aware of a way to turn it off.