sidoh/esp8266_milight_hub #757: Update platform version to possibly fix wifi issue in espressif8266

iDont, 2021-12-23:
Thank you for esp8266_milight_hub. It is working perfectly for me!

However, it can only connect to my 2.4G wifi network if I disable the HT LDPC coding capability of the radio device. It seems there was a bug in the espressif8266 platform that was fixed a while ago: https://github.com/kaloz/mwlwifi/issues/278#issuecomment-703139315

Is it possible to update the platform in plaform.ini so hopefully connecting to certain wifi routers will be a bit easier?

andiaa734, 2022-01-09:
I'm currently on Platform v2.6.3.

But for that you have to change MiLightClient.h:

`#define FS(str) (reinterpret_cast<const __FlashStringHelper*>(str))
`

to

`#define FSH(str) (reinterpret_cast<const __FlashStringHelper*>(str))
`

and all occurrences of FS in MiLightClient.cpp to FSH.

Thats necessary because FS is introduced in FS.h for Filesystem.

andiaa734, 2022-01-09:
Following Updates I'm currently testing:

RF24@~1.3.2 to 1.4.2
  ArduinoJson@~6.10.1 to 6.16.0
  PubSubClient@~2.7 to 2.8.0
  https://github.com/ratkins/RGBConverter.git#07010f2
  WebSockets@~2.2.0
  CircularBuffer@~1.2.0 to 1.3.3
  PathVariableHandlers@~2.0.0
  RichHttpServer@~2.0.2

For ArduinoJson 6.19.0 there are code changes required to MiLightClient.cpp as the Method isUndefined() was an unoffical Method an has been removed. I have replaced isUndefinde() with isNull() but I'm not sure if the meaning is the same and isNull() is exactly doing what @sidoh wants to do in the code.