sidoh/esp8266_milight_hub #758: Milighthub on Feather Huzzah "forgets" configuration

fieldOfView, 2021-12-29:
### Describe the bug

After more than a year of service, my milight-hub instance running on a Feather Huzzah seems to have become "forgetful". In the span of a day or less, all settings except the wifi config and the theme will reset to the defaults. The `/settings` endpoint returns a valid but "default" json document, and the interface has lost my aliases. Restoring a backup gets everything back to working order, until the configuration is forgotten again. 

This behavior started happening last week, and is now an approximately daily occurrence.

### Steps to reproduce

Sofar I have not been able to reliably reproduce the issue. Restarting via the webinterface or power cycling the board does not trigger the forgetfulness.

After a couple of hours the configuration will be reset

### Expected behavior

The configuration is remembered.

### Setup information

Milighthub on Feather Huzzah (8266) with a NRF24L01+ radio.

#### Firmware version
Current Version: 1.10.8 (huzzah)
Issue started happening with 1.10.7 (huzzah), upgrading to 1.10.8 did not change it.

#### Output of http://milight-hub.local/about

```json
{

    "firmware": "milight-hub",
    "version": "1.10.8",
    "ip_address": "10.0.2.9",
    "reset_reason": "External System",
    "variant": "huzzah",
    "free_heap": 16704,
    "arduino_version": "2_4_2",
    "queue_stats": {
        "length": 0,
        "dropped_packets": 0
    }

}
```

#### Output of http://milight-hub.local/settings

```json

{
    "admin_username": "***",
    "admin_password": "***",
    "ce_pin": 4,
    "csn_pin": 15,
    "reset_pin": 0,
    "led_pin": -2,
    "radio_interface_type": "nRF24",
    "packet_repeats": 50,
    "http_repeat_factor": 1,
    "auto_restart_period": 0,
    "mqtt_server": "10.0.2.4",
    "mqtt_username": "***",
    "mqtt_password": "***",
    "mqtt_topic_pattern": "milight/command/:device_id/:device_type/:group_id",
    "mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
    "mqtt_state_topic_pattern": "milight/state/:device_id/:device_type/:group_id",
    "mqtt_client_status_topic": "milight/status",
    "simple_mqtt_client_status": false,
    "discovery_port": 48899,
    "listen_repeats": 3,
    "state_flush_interval": 10000,
    "mqtt_state_rate_limit": 500,
    "mqtt_debounce_delay": 500,
    "mqtt_retain": true,
    "packet_repeat_throttle_sensitivity": 0,
    "packet_repeat_throttle_threshold": 200,
    "packet_repeat_minimum": 3,
    "enable_automatic_mode_switching": false,
    "led_mode_wifi_config": "Fast toggle",
    "led_mode_wifi_failed": "On",
    "led_mode_operating": "Slow blip",
    "led_mode_packet": "Flicker",
    "led_mode_packet_count": 3,
    "hostname": "BabbersLichtjes",
    "rf24_power_level": "MAX",
    "rf24_listen_channel": "LOW",
    "wifi_static_ip": "",
    "wifi_static_ip_gateway": "",
    "wifi_static_ip_netmask": "",
    "packet_repeats_per_loop": 10,
    "home_assistant_discovery_prefix": "",
    "wifi_mode": "n",
    "default_transition_period": 500,
    "rf24_channels": [
        "LOW",
        "MID",
        "HIGH"
    ],
    "device_ids": [
        22408
    ],
    "gateway_configs": [ ],
    "group_state_fields": [
        "state",
        "brightness",
        "mode",
        "color_temp",
        "bulb_mode",
        "computed_color"
    ],
    "group_id_aliases": {
        "Keuken": [
            "fut091",
            22408,
            4
        ],
        "Raamkant": [
            "fut091",
            22408,
            1
        ],
        "Tafel": [
            "fut091",
            22408,
            3
        ],
        "TomDixon": [
            "fut091",
            22408,
            2
        ]
    }
}
```

### Additional context

fieldOfView, 2022-01-02:
Output of http://milight-hub.local/about after forgetting:
```
{

    "firmware": "milight-hub",
    "version": "1.10.8",
    "ip_address": "10.0.2.9",
    "reset_reason": "External System",
    "variant": "huzzah",
    "free_heap": 19120,
    "arduino_version": "2_4_2",
    "queue_stats": {
        "length": 0,
        "dropped_packets": 0
    }

}
```

Output of http://milight-hub.local/settings after forgetting
```
{

    "admin_username": "",
    "admin_password": "",
    "ce_pin": 4,
    "csn_pin": 15,
    "reset_pin": 0,
    "led_pin": -2,
    "radio_interface_type": "nRF24",
    "packet_repeats": 50,
    "http_repeat_factor": 1,
    "auto_restart_period": 0,
    "mqtt_server": "",
    "mqtt_username": "",
    "mqtt_password": "",
    "mqtt_topic_pattern": "",
    "mqtt_update_topic_pattern": "",
    "mqtt_state_topic_pattern": "",
    "mqtt_client_status_topic": "",
    "simple_mqtt_client_status": false,
    "discovery_port": 48899,
    "listen_repeats": 3,
    "state_flush_interval": 10000,
    "mqtt_state_rate_limit": 500,
    "mqtt_debounce_delay": 500,
    "mqtt_retain": true,
    "packet_repeat_throttle_sensitivity": 0,
    "packet_repeat_throttle_threshold": 200,
    "packet_repeat_minimum": 3,
    "enable_automatic_mode_switching": false,
    "led_mode_wifi_config": "Fast toggle",
    "led_mode_wifi_failed": "On",
    "led_mode_operating": "Slow blip",
    "led_mode_packet": "Flicker",
    "led_mode_packet_count": 3,
    "hostname": "milight-hub",
    "rf24_power_level": "MAX",
    "rf24_listen_channel": "LOW",
    "wifi_static_ip": "",
    "wifi_static_ip_gateway": "",
    "wifi_static_ip_netmask": "",
    "packet_repeats_per_loop": 10,
    "home_assistant_discovery_prefix": "",
    "wifi_mode": "n",
    "default_transition_period": 500,
    "rf24_channels": [
        "LOW",
        "MID",
        "HIGH"
    ],
    "device_ids": [ ],
    "gateway_configs": [ ],
    "group_state_fields": [
        "state",
        "brightness",
        "computed_color",
        "mode",
        "color_temp",
        "bulb_mode"
    ],
    "group_id_aliases": { }

}
```

ml9907, 2022-01-03:
Same here with nodemcu. It was a daily routine to upload the config json for a while. I've switched to D1mini and the problem solved.
1.10.7 & 1.10.8 firmware downloaded and locally built gave the same problem on nodemcu. Now the D1mini has downloaded 1.10.8 and remained intact since more than a week.

fieldOfView, 2022-01-30:
I have also since replaced the Feather Huzzah with a more compact Wemos mini D1. 

My hunch (with nothing to back that up) is that perhaps milight hub - under certain circumstances - writes to the configuration too often which leads to occasional corruption. That would explain why the forgetfulness started happening after more than a year of working just fine and also why switching to a new microcontroller fixes the issue.