sidoh/esp8266_milight_hub #762: 5in1 Controller to use in Single Color Mode

MaAnMa-manuel, 2022-01-09:
Hi!
First. GREAT PROJECT! Really love the light weight light solution!

### What is the model number of the device you're trying to control?
Model No.: LS2
5 IN 1 Smart LED Controller

### What firmware version(s) have you tried? 
1.10.8 (d1_mini)

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
D1_Mini

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGBW mode of the controller works fine!
How to use the singe mode I don't understand. 

### Have you tried controlling the device with a physical remote?
The Device works fine. Only the Single Mode?

Thank you for your help!
BR, Manuel

markg85, 2023-01-18:
I too am curious if this is supported?

While i hate milight with their stock - very bad - hub and app, using this very project makes it a super powerful awesome platform!

philletourneau, 2025-02-10:
I have the same issue with single color mode