sidoh/esp8266_milight_hub #763: FUT012 won't pair?

MisterC925, 2022-01-19:
It's been a few years but one of my bulbs went out so I bought a FUT012 "MIBOXER" branded 9W bulb.  

The FUT012 is listed as being RGB+CCT on the wiki so I attempted to pair by setting the id (I've tried several, most recently Hub 1 ID 0x2022) and hitting pair within 5 seconds of applying power.

The bulb doesn't blink nor does it respond to on/off or other commands.

I fully admit I haven't done this since the initial setup ages ago but I think I'm following the correct procedures?

I did a capture and the only traffic I see during the paring attempt is variations of:
rgb_cct packet received (9 bytes):
Raw packet: 00 DB C1 01 66 D1 A6 66 E8 

Decoded:
Key      : 00
b1       : 20
ID       : 2022
Command  : 01
Argument : 01
Sequence : 14
Group    : 01
Checksum : 31

with different sequence and checksum #'s.

All of my existing lights are responding so the radio is working.

Any ideas?  It it possible MiBOXER FUT012 is a different bulb? I can't really see it since MiBoxer is a Mi-light subbrand (according tot he box?) 

I don't know where my remotes are to double-check the bulb is working but the box lists the following as compatible:
FUT089/089-B
FUT092/092-B
FUT088
FUT096/096-B
S2-B/S2-W/S2-G

the wiki indicates RGB+CCT (FUT092) settings should work, I also tried FUT089/B8 and it does the same.

Any Ideas?

Many Thanks!

MisterC925, 2022-01-26:
So I found the PDF of the user manual and the instructions for pairing are:
Short press " I " button 3
times within 3 seconds
when light on

This seems different than I remember with the other bulbs.  I've tried hitting "Pair" 3 times in the GUI but that doesn't work.

Thanks!
-S
http://futlight.com/light/m_n/pdf/RGBCCT_Lighting_2_4G_RF/RGBCCT_Lighting_EN_V1.0.pdf

MikeCLX, 2022-02-02:
I don’t have a fut012 sorry but have tried a heap of other ones with RGB+CCT as the remote and just click pair once within a few seconds of applying power to the bulbs and the light should flash green a few times. Every different RGB+CCT bulb or light I have tried has worked first time.

the “I” 3 times is just for the remote, have you got it paired to a remote? Maybe try unpairing it from that first.

MikeCLX, 2022-02-02:
Or try using 0xE68 that’s the code of a real RGB-CCT remote and works for me

MisterC925, 2022-02-02:
I have probably 6 other RGB+CCT bulbs and they all worked easy enough out of the box. 

Unfortunately, any remotes I have that are packed away since I haven't needed them for a couple of years at least.  

The FUT012 in question is the first one I got with "MiBOXER"  branding ("Mi-Light Subordinate New Brand" is also on the box) instead of just Mi-Light so that is why I was wondering if the pairing procedure was different. 

I tried 0xE68 as device name with no luck. Is there a shorter distance for pairing (like Zigbee/Zwave) for the bulbs, so I could put in in a fixture close to the hub and then move it back to it's final location?

MikeCLX, 2022-02-02:
I Should have said,  mine are all less than a month old miboxer branded, so procedure has not changed, yes I would try it close maybe the pair command is not being repeated by the others to get to your bulb