sidoh/esp8266_milight_hub #764: FUTCX range

electricfire, 2022-01-23:
hi will this support the FUTCx range?  I tried to click on the supported bulbs but i get a 404

MikeCLX, 2022-02-02:
I just connected a FUTC05 and it worked perfectly, what part number do you have?

electricfire, 2022-02-08:
Ok that is good to know i haven't got it yet just could not see them in the page of supported bulbs, probably be FUTC05/03/04.  Also on a side note when you are pairing the bulbs together does it have to be same type i tried to pair two different bulbs and only one would pair.

MikeCLX, 2022-02-08:
I have a number of different types but have not tried that yet, I can do though, that said as the bulbs are just receiving a signal to pair with they would have no idea that the other bulb existed, some bulbs need to be unpaired before pairing and they do seem to need to be close, I don’t think they repeat the pair signal. I’ll be able to try it in a few days.

electricfire, 2022-02-08:
Yea that would be good if you can confirm, i have the two bulbs next to each other, when you unpair you don't get notification that it has unpaired, so just just to be sure I deleted the alias and the device ID to confirm and went through the pairing again, interestingly they both flashed during the pairing but when i tried to control only one of them would respond, FYI i was trying to pair a FUT106 and FUT103

electricfire, 2022-02-25:
@MikeCLX did you have any luck testing it out?