sidoh/esp8266_milight_hub #765: Unable to restore from back-up

cordvision, 2022-01-24:
I’m running into a very strange issue with my hub and I was wondering if you might be able to help.
All of the sudden, my hub was reset to the default values.
I then tried to restore from my back-up, but even though I got the “success” message, the hub was still in the “factory setting”
state. 
After spending a lot of time experimenting with the back up file, I identified several mqtt parameters that appear to be the cause.
I marked them in bold bellow. If I leave those parameters empty, I can restore the hub, but if I try to restore with those parameters as shown in bold bellow, the hub just resets to the default values (factory settings).

If I restore the hub with the back-up file that has the parameters marked in bold edited out, I can then add those parameters again in the UI (in the MQTT tab). The hub will then run fine until it gets disconnected from power or rebooted. If that happens, it will return to factory settings.

I also downgraded the firmware, to the previous version, but the issue persists.

Any idea what could cause this?

Best
Michael

`
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":250,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.2.167:1883","mqtt_username":"homeassistant","mqtt_password”:"testpassword22254323",

_**"mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/state/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status",**_

"simple_mqtt_client_status":true,"discovery_port":
48899,"listen_repeats":10,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":100,"packet_repeat_minimum":250,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"HIGH","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":125,

_**"home_assistant_discovery_prefix":"homeassistant",**_

"wifi_mode":"n","default_transition_period":250,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[14617,14624,14625,14626,14627,14628,14629,14630,14631,14632,14633,14640,14641,14642,14643,14644,14645,18844,20481,48385],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"All_Garden":["fut089",18844,8],"Bed Ceiling 1":["rgbw",14632,1],"Bed Ceiling 2":["rgbw",14633,1],"Bed Ceiling 3":["rgbw",14640,1],"Bed Corner Lamp 1":["rgbw",14629,1],"Bed Corner Lamp 2":["rgbw",14630,1],"Bed Corner Lamp 3":["rgb_cct",14631,1],"Bed Lamp L 1":["rgbw",14625,1],"Bed Lamp L 2":["rgbw",14626,1],"Bed Lamp R 1":["rgbw",14627,1],"Bed Lamp R 2":["rgbw",14628,1],"Bed Strip":["rgb_cct",14624,1],"Corner Lamp":["rgbw",48385,1],"Corridor":["rgb_cct",14617,2],"Living Room Ceiling":["rgbw",14641,1],"Living Room Corner Lamp 1":["rgbw",14642,1],"Living Room Corner Lamp 2":["rgbw",14643,1],"Living Room Corner Lamp 3":["rgbw",14644,1],"Living Room Corner Lamp 4":["rgbw",14645,1],"Office":["rgb_cct",14617,1]}}
`

alkhankhel, 2022-02-01:
I have same problem, when settings.json > ~2.2kB
If you set it up again, then after a reboot everything will be reset.

VDRainer, 2022-05-13:
Same problem here.
Made a backup (1.9kB) and restartet the hub (wemos d1 mini)
After restart everything reset.
Not possible to restore backup.

starsoftnl, 2022-09-24:
allmost same problem here (nodeMCU). For me it works after removing te last device from the json file. After putting it back restore does not work.

[settings all formatted.zip](https://github.com/sidoh/esp8266_milight_hub/files/9639595/settings.all.formatted.zip)

DigiH, 2023-07-08:
I have also hit the >**2.2kB** limit for settings.json when trying to add some new lights to my home the last few days. Clearly indicated by the serial output

> Error parsing saved settings file: NoMemory

So what I did was to shorten the Device Names manually in settings.json to more cryptic shorter strings, to get it under 2.2kB again, and all is fine - also across restarts, which I have set up in a regular interval auto_restart_period for other reasons.

With even more lights coming in the future I am wondering what the best way forward is.

One option would also be to shorten many of the very long internal variable names stored in settings.json in my custom build, but any other suggestions are more than welcome.

DigiH, 2023-07-09:
Ok, I've bumped **MILIGHT_HUB_SETTINGS_BUFFER_SIZE** up to 6144 (from the previous default 4096) for now and all is fine even with my longer Device Names and newly added bulbs. So no more >2.2kB limit - no more 

> Error parsing saved settings file: NoMemory

when 

```
…
    DynamicJsonDocument json(MILIGHT_HUB_SETTINGS_BUFFER_SIZE);
    auto error = deserializeJson(json, f);
…
```

Hope this helps others with the same problem.

sidoh, 2023-09-27:
This should hopefully be resolved in v1.11.0. I split aliases out of the settings file.