sidoh/esp8266_milight_hub #769: Reboot at specific times

smarticz, 2022-02-01:
Would it be possible to add in the settings, automatic restart of the device?

Sometimes it gets lags and only rebooting helps.

For example:
- Never
- Every hour
- Every 3 hours
- Every 6 hours
- Every 12 hours
- At 4 am

garmck, 2022-02-01:
This is already implemented.
Settings / Setup / Auto-restart period

VDRainer, 2022-05-15:
You can also use curl to restart the hub at specific times.
For example with a homeassistant shell_command.
```
# Restart Milight Hub
shell_command:
  mlh_eg_restart: "curl http://192.168.2.62/system -X POST -H 'Content-Type: application/json' -d '{\"command\":\"restart\"}'"
```
But NOT if you have this https://github.com/sidoh/esp8266_milight_hub/issues/765 issue. :neutral_face:

DigiH, 2023-07-09:
@VDRainer 

Also possible across restarts with my comment in https://github.com/sidoh/esp8266_milight_hub/issues/765#issuecomment-1627745981