sidoh/esp8266_milight_hub #772: wemos mini d1 v4 keeps rebooting

sedlacekdavid, 2022-03-05:
Hi, my problem is that wemos keeps restarting (at least I think it is happening). I see that the led is bliping (not as fast as like it is in the wifi config mode). Even though this LED bliping is not configured. I can't save any settings, no matter what I put in and hit Save, then after reloading the page I will find every field empty. When I click on Backup download, nothing happens.

I have latest Wemos D1 mini v4 (the one with a USB C port). Tried to flash prebuilt [1.10.8] hexes (almost all of them). Effect is the same. It has ESP8266EX chip with 4MB onboard flash.

I tried different power sources and running only the bare board without having anything connected to it.

It is even stranger that there are no default values when I open the config page.

Any suggestions? Thank you very much.

[image]

sedlacekdavid, 2022-03-05:
�en�Hd�x$Z�Error parsing saved settings file: IncompleteInput
*WM: Adding parameter
*WM: staticIP        
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.1.164
Wifi connected succesfully

Setup complete (version 1.10.8)

sedlacekdavid, 2022-03-06:
Same issue using a different board (same model, a brand new one as well).
This is the output of the fresh unit:

```

> Executing task in folder esp8266_milight_hub: C:\Users\David.Sedlacek\.platformio\penv\Scripts\platformio.exe device monitor --environment d1_mini <

--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on COM7  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�EO,�`�l$�␆xh�␐��*WM: Adding parameter
*WM: staticIP        
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
*WM:        
*WM: AutoConnect       
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM:
*WM: Configuring access point... 
*WM: ESP14527539
*WM: milightHub
*WM: AP IP address: 
*WM: 192.168.4.1        
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Parameter
*WM: staticIP
*WM:
*WM: Parameter
*WM: netmask
*WM: 
*WM: Parameter
*WM: gateway
*WM:
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Wifi connected succesfully
[... 836 lines cut ...]
3fffffa0:  00000000 00000000 3fff19de 40209666  
3fffffb0:  3fffdad0 00000000 3fff19de 4022bfc0  
3fffffc0:  feefeffe feefeffe 3ffe872c 40100721  
<<<stack<<<
H!�1'D�
␡Error parsing saved settings file: IncompleteInput
*WM: Adding parameter
*WM: staticIP
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.165
Wifi connected succesfully

Setup complete (version 1.10.8)

````

Ninja1283, 2022-11-04:
Same issue here with three brand new v4 boards. I couldn't even get to the WiFiManager page to connect the board to my AP before a reboot event. My previous ESP12F board worked fine; I just wanted to standardize on USB-C connections.

nicofamdj, 2023-05-19:
Same issue here.

sidoh, 2023-09-27:
I don't have any v4 boards to test with, but I just ordered some.

If you haven't already, perhaps try clearing flash with esptool and reflashing?

sidoh, 2023-10-01:
Just tried flashing and it seems to have worked without any fiddling. I did flash the new version I just released so possible (although unlikely) that something in one of the SDK upgrades fixed things.

Gonna close since I can't replicated but please feel free to comment/re-open if still occurring!

closed 2023-10-01 by sidoh

sedlacekdavid, 2023-10-06:
Hi, first of all, thank you for keeping this alive! 
I thought it is already a dead project (from new development perspective) - although I've been using it for 2 years!

I gave it a try and flashed exactly the same Wemos D1 mini v4 board as year ago - and it works!

So thx a lot and keep it up, David