sidoh/esp8266_milight_hub #773: JSON Output for Home Assistant

poudenes, 2022-03-29:
The MQTT Light config in Home Assistant has been changed/improved. 
https://www.home-assistant.io/integrations/light.mqtt/

They use follow JSON Schema:

{
  "brightness": 255,
  "color_mode": "rgb",
  "color_temp": 155,
  "color": {
    "r": 255,
    "g": 180,
    "b": 200,
    "c": 100,
    "w": 50,
    "x": 0.406,
    "y": 0.301,
    "h": 344.0,
    "s": 29.412
  },
  "effect": "colorloop",
  "state": "ON",
  "transition": 2,
}

Thats different then Milight_hub sends. 
Im not technical enought to understand the difference, but it seems there is a mismatch...

Can you look into this?

poudenes, 2022-10-24:
For example:
**HS Color (hue)**
```
mqtt:
  light:
    - schema: json
      name: mqtt_json_hs_light
      state_topic: "home/light"
      command_topic: "home/light/set"
      color_mode: true
      supported_color_modes: ["hs"]
```
HA expect:
```
{
  "state": "ON",
  "color_mode": "hs",
  "color": {
    "h": 24.0,
    "s": 100.0
  }
}
```

devax, 2022-11-22:
Sorry for piggybacking, but my problem seems somehow related.
Home Assistant is pushing me to update my manual configuration that is currently under 
```yaml
light:
  platform: mqtt
  ...
```
to the new format which should be
```yaml
mqtt:
  light:
    ....
```
The wiki seems to be outdated, as it still shows the legacy configuration (which seems to be unsupported with the upcoming version 2022.12): https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant#hass

Can I find somewhere a working example for the new configuration?

My confusion starts when trying to setup the new configuration in VS Code, as it's giving me a lot of red lines:
[image]

I would appreciate if someone could point me to a good starting guide or to a working example.

Or is this new configuration not supported by milight hub?

poudenes, 2022-11-22:
> Sorry for piggybacking, but my problem seems somehow related. Home Assistant is pushing me to update my manual configuration that is currently under
>
> ```yaml
> light:
>   platform: mqtt
>   ...
> ```
>
> to [...]

To help you out (offtopic of this whole issue):

this is how I did setup

mqtt:
  light:
    - name: "Balcony All"
      unique_id: "2b23416c-0030-4d99-af4d-99b8d8cf78ad"
      command_topic: "milight/0xAAA4/rgb_cct/1"
      state_topic: "milight/states/0xAAA4/rgb_cct/1"
      schema: "json"
      availability_topic: "milight/client_status"
      payload_available: "connected"
      payload_not_available: "disconnected"
      brightness: true
      effect: true
      effect_list: ["night_mode","white_mode","0","1","2","3","4","5","6","7","8"]
      rgb: true
      color_temp: true
      max_mireds: 370
      min_mireds: 153
      retain: false
      optimistic: true

closed 2023-04-13 by poudenes