sidoh/esp8266_milight_hub #775: Bugs in LinkedList.h

juupje, 2022-08-10:
### Describe the bug
The function `ListNode<T>* LinkedList<T>::getNode(int index)` will always return false. This is due to a missing statement that returns `current`.
Moreover, the function `LinkedList<T>::add(int index, T _t)` does not assign the `prev` pointer to the object that is to be inserted.

### Steps to reproduce

This seems to be of no consequence because `getNode` is not used outside `LinkedList.h` and neither are any methods that depend in it. Also, the `prev` pointer in ListNode is not used outside of `LinkedList.h`.

### Expected behavior

Even though these bugs are currently of no consequence for the existing code, they cause issues for an alarm feature that I am attempting to implement and might cause unexpected behavior in the future.

The correct return statement of `getNode` should be:
```C++
  return _pos==index ? current : false;
```
and the current code of `add` include `tmp->prev = _prev;`.

### Setup information

#### Firmware version
1.10.8

Additional information about my setup is not necessary

sidoh, 2022-08-10:
whoops. good catch! if you wanna throw up a PR with these changes happy to merge.

juupje, 2022-08-11:
I created a pull request (I think; I'm a newby to git so apologies if I messed up).
And many thanks for your efforts on creating this project!

closed 2022-08-11 by juupje