sidoh/esp8266_milight_hub #780: MQTT Light in Home Assistant 2022.10.x

poudenes, 2022-10-23:
Hi All,

I have some problems with MQTT lights and Home Assistant when MQTT Broker is for a moment offline and come online again.
When this happen some lights went on and others keep off. I really don't know how to solve this.

I cleared MQTT broker one time but didn't work. Retain is off but its getting me crazy. 
Below the config how I add the lights into HA:

```
light:
  - name: "Livingroom All"
    unique_id: "68ee8212-e45a-4898-aa66-42d7f16358e8"
    command_topic: "milight/0xAAAC/rgb_cct/0"
    state_topic: "milight/states/0xAAAC/rgb_cct/0"
    retain: true
    <<: &MILIGHT_PARAMS
      schema: "json"
      availability_topic: "milight/client_status"
      payload_available: "connected"
      payload_not_available: "disconnected"
      brightness: true
      effect: true
      effect_list:
        - "night_mode"
        - "white_mode"
        - "0"
        - "1"
        - "2"
        - "3"
        - "4"
        - "5"
        - "6"
        - "7"
        - "8"
      rgb: true
      color_temp: true
      max_mireds: 370
      min_mireds: 153
```

Because some lights are in a group (all) that's /0 instead of /1, /2, /3 or /4 for the group. I use this to patch all separate groups with same topic information (to sync Milight hub with HA)

```
- id: "0596ec6b-233a-4e44-a107-51004000f38a"
  alias: "Patches - MiLight Livingroom Group 0"
  mode: restart
  initial_state: true
  trigger:
    - platform: mqtt
      topic: "milight/states/0xAAAC/rgb_cct/0"

  condition: []

  action:
    - service: mqtt.publish
      data:
        topic: "milight/states/0xAAAC/rgb_cct/1"
        payload_template: "{{ trigger.payload }}"

    - service: mqtt.publish
      data:
        topic: "milight/states/0xAAAC/rgb_cct/2"
        payload_template: "{{ trigger.payload }}"

    - service: mqtt.publish
      data:
        topic: "milight/states/0xAAAC/rgb_cct/3"
        payload_template: "{{ trigger.payload }}"

    - service: mqtt.publish
      data:
        topic: "milight/states/0xAAAC/rgb_cct/4"
        payload_template: "{{ trigger.payload }}"

    # LIVINGROOM WALL WINDOWS BULBS
    - service: mqtt.publish
      data:
        topic: "milight/states/0xAABB/rgb_cct/1"
        payload_template: "{{ trigger.payload }}"

    - service: mqtt.publish
      data:
        topic: "milight/states/0xAABB/rgb_cct/2"
        payload_template: "{{ trigger.payload }}"
```

So when I turn on or off or change color, kelvin in group 0 all other groups 1,2,3 and 4 also changed nicely.
Now you know the setup the question:

When I turn off all lights and the MQTT Broker goes offline and online again some lights go on. (look like the status is still on in MQTT broker)

I hope someone understand what I mean....

poudenes, 2022-10-24:
```
Here my MQTT settings
{
	"mqtt_topic_pattern": "milight/:hex_device_id/:device_type/:group_id",
	"mqtt_update_topic_pattern": "",
	"mqtt_state_topic_pattern": "milight/states/:hex_device_id/:device_type/:group_id",
	"mqtt_client_status_topic": "milight/client_status",
	"mqtt_state_rate_limit": 501,
	"mqtt_debounce_delay": 500,
	"mqtt_retain": false,
	"simple_mqtt_client_status": true,
	"enable_automatic_mode_switching": false
}
```

sidoh, 2022-10-24:
I don't have a ton of time to look into this in detail unfortunately, but if lights are changing state when your MQTT broker restarts, it's likely the case that you have some old retained messages laying around. If you don't care about any retained messages your MQTT broker might have, you could try just clearing its storage. Otherwise you can probably poke around with an MQTT client like `mosquitto_pub` / `mosquitto_sub`. I believe publishing a retained message with an empty body is how you delete retained messages in MQTT.

poudenes, 2022-10-24:
Thanks let me check. I removes today all retain: true in my HA config and removed the option in milight hub. But when I restart Ha again all lights become unavailable.... I need to turn them on and off in the hub to make them working again.

poudenes, 2022-10-24:
that's also what I see when I do a manual reboot. Then connect with MQTT Explorer and see all the states.
The lights are all configured correct as you could see in my earlier post. For now I have enable retain again and now when reboot HA all lights are back as normal. But without retain it seems to fail. Maybe a bug inside HA version.... IDK

Linkenelis, 2022-10-28:
I guess it is a similar issue I have...
When you give an MQTT cmnd, it stays there. You have to remove the command in MQTT cmnd after 1 second so that after a reconnect (or device reset), the command isn't repeated.
My broker will not remove commands given...

Linkenelis, 2022-11-02:
Hmm 
I use iobroker, so something similar.
There is a script running that whenever this value (eg: milight/cmnd/0x01/01) changes, 500 ms (or 1 sec) later it will be written with text ""
so in your case in HA, as that is where the command is?

poudenes, 2022-11-02:
After some hard thinking couple of days, it guess found the issue. Im make use of ESP Easy wall switches. Custom made by "Meek" There is a touch button. When I reboot this ESP the touch went on so it activates rules inside the ESP.

Now find out that I can send a "Send boot state". Since then if I restart the ESP the light stay off.
Then I checked other ESP Wall switches where the light already stayed off. And they had already turned on this "Send boot state".

Then I checked my last one this will turn on a RFlink dimmer and always went on after ESP restart or when its disconnect from Wifi and connect again. Also enabled this "Send boot state" and also this lights now stay off..

I keep this in scope. But seems that my problem was all the time something else... :)

Linkenelis, 2022-11-03:
:-)

closed 2023-04-13 by poudenes