sidoh/esp8266_milight_hub #786: On MH Platform via Homebridge 2019 Mi-Boxer bulbs work great! 2020 Mi-Boxer bulbs do not.

mames13, 2023-01-16:
Hi, I have gone through the trouble shooting guide, thank you for posting that.  

### What is the model number of the device you're trying to control?

FUT014 6 watt (some purchased in 2019, some purchased in 2020)
FUT012 9 watt (some purchased in 2019, some purchased in 2020)
B8 Remote / Wall Panel 

### What firmware version(s) have you tried? 

esp8266_milight_hub_esp12-1.10.8.bin

I wasn't able to figure out PerformIO (please let me know if there is a step by step online) so I used these Bash commands:

`python3 -m esptool write_flash 0 /Users/mamiemccall/Downloads/esp8266_milight_hub_esp12-1.10.8.bin`

and

`esptool.py --chip esp8266 --port /dev/cu.usbserial-0001 erase_flash`

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 

Nodemcu v2 
HiLetgo (amazon)

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 

So far I have been selecting the B8 Remote FUT089. I think that remote is considered RGB+CCT as well. Have been wondering if the two (in the web config UI) are interchangeable? 

### Have you tried controlling the device with a physical remote?

Yes! The bulbs will control with The B8 remotes and with the Esp MH web UI perfectly. 

The problem is, the MiBoxer bulbs that I bought in 2019 can only be paired with the bulbs I bought from that year (maybe bulbs from before 2019 would work with the 2019's as well) but if I try to add the MiBoxer bulbs that I bought in 2020 to the same remote and channel as the 2019's the colors will not sync. This has always been an issue before starting the Esp Milight Hub project so I just make sure to keep the new bulbs on different remotes from the old.  

I am doing the same thing when I group the bulbs vs remotes in the Esp MH web UI.   

The bulbs I purchased in 2019 all work great with the Esp and Homebridge Platform (including Siri commands) but the bulbs from 2020 do not. The 2020 bulbs work maybe 2 time before seemingly timing out in Home or when using Siri (connected through Homebridge).  The continue to work however in the web UI. 

 

B8 FUT089 (remote of choice / have many of these)

S2-W, new, small, round, 1 group remote with RGB+CCT and brightness fade. (I just bought one of these to test out) 

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
"... /about output. put between the ```s"
```

`{"firmware":"milight-hub","version":"1.10.8","ip_address":"192.168.1.5","reset_reason":"Software/System restart","variant":"esp12","free_heap":17504,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}`

#### /settings

```json
"... /settings output. put between the ```s"
```

`{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":2,"auto_restart_period":20,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":6,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"On","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.168.1.5","wifi_static_ip_gateway":"192.168.1.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","HIGH"],"device_ids":[16388,16392,24577],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Closet":["fut089",16392,2],"Echo":["fut089",24577,1],"Globe-East":["fut089",16388,1],"Globe-West":["fut089",16388,1]}}`

The bulbs called, "Globe-East", "Globe-West", and "Closet" were all purchased in 2019 and work perfectly at all times in both the Web UI and in Homekit / with Siri commands. The bulb called "Echo" is from 2020 and does not work in Homekit (only a couple of times before Homekit stops being able to control it). I have tried others from 2020 (calling them Alpha, Beta, Delta and assigning them different device ID's) and they all don't work reliably either. 

HOMEBRIDGE CONFIG (In case this helps) :
{
    "host": "192.168.1.5",
    "name": "MiLightHubPlatform",
    "backchannel": true,
    "darkMode": false,
    "rgbcctMode": true,
    "forceHTTP": false,
    "syncHubInterval": 5,
    "commandDelay": 100,
    "debug": true,
    "_bridge": {
        "username": "0E:2C:5C:E1:26:EE",
        "port": 52381
    },
    "platform": "MiLightHubPlatform"
}

Thank you!

mames13, 2023-07-06:
I think the problem has to do with the fact that I switched to a wifi 6 access point. Still having problems.

domoticaDude, 2023-12-29:
> I think the problem has to do with the fact that I switched to a wifi 6 access point. Still having problems.

temporary disable Wifi 6 and enable only 2g. Most smart devices have problems with wifi6. When it works you can enable Wifi6 again. And it should work fine. If  wifi 6 is the problem.