sidoh/esp8266_milight_hub #793: Cannot make Mi-Light Series - CCT working

FilipDem, 2023-05-30:
Hi all,
I am really getting headache... I am rather an experienced technician/ITer, but doesn't get this working.
Let me summarize the reason of my headache:
* I have a wireless dimmer as indicated below used together with a FUT007 (CCT) remote control (see also picture below). Both works fine together.
[image]
[image]

* I am using a ESP8266 Nodemcu with NRF24L01 connected (see detailed information below). Both are temporarily wired with short jumper/Dupont cables. Sniffing works... When I start the sniffing and I press a button on the remote, I got
```
cct packet received (7 bytes)
Request type : 5A
Device ID       : E688
b1                  : 01
b2                  : 0B
b3                  : 60
Seq.num.       : 3B
```
* When I use the Mi-Light ESP8266 Hub, I am using the same DeviceID (0XE688), Group 1 and Remote Type CCT. Clicking ON/OFF doesn't give any effect. 
* As I use the same DeviceID, I assume that no pairing is required (I am just using the same data as the remote). However after having unpaired with the remote, I am trying to pair with the Mi-Light EPS8266 Hub (after powering off/on the wireless dimmer and within 5 seconds): nothing happens, ...
* In the sniffing area I got anyhow the correct information (same DeviceID, b1 corresponding to Group 1 etc). I know that the information sent by the Mi-Light ESP8266 Hub is not really "sniffed", but I assume that sending works (seen the sniffing of the remote works).
* I already bought a new NRF24 following the recommendation in the troubleshooting... same problem...
* As the receiving works, I assume the sending should also work... I checked the connectivity several times, changed jumper wires, added a capa 10µF etc... I cannot get it working.

Is there anything I might have overlooked? I already investigated the troubleshooting... The only thing I didn't try is using a second Mi-Light HUB to snif eachother (I don't have a second Nodemcu available at this moment). But again, assuming sending works as I can receive information.

Thanks for help!

### What is the model number of the device you're trying to control?
[image]

### What firmware version(s) have you tried? 
1.10.8 (nodemcuv2)

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
nodemcu

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
CCT

### Have you tried controlling the device with a physical remote?
Yes. 
[image]

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```
{
  "firmware": "milight-hub",
  "version": "1.10.8",
  "ip_address": "192.168.1.61",
  "reset_reason": "Power on",
  "variant": "nodemcuv2",
  "free_heap": 18424,
  "arduino_version": "2_4_2",
  "queue_stats": {
    "length": 0,
    "dropped_packets": 0
  }
}
```

#### /settings
```
{
  "admin_username": "",
  "admin_password": "",
  "ce_pin": 4,
  "csn_pin": 15,
  "reset_pin": 0,
  "led_pin": -2,
  "radio_interface_type": "nRF24",
  "packet_repeats": 4,
  "http_repeat_factor": 2,
  "auto_restart_period": 0,
  "mqtt_server": "",
  "mqtt_username": "",
  "mqtt_password": "",
  "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
  "mqtt_update_topic_pattern": "milight/updates/:hex_device_id/:device_type/:group_id",
  "mqtt_state_topic_pattern": "milight/states/:hex_device_id/:device_type/:group_id",
  "mqtt_client_status_topic": "milight/connection_status",
  "simple_mqtt_client_status": false,
  "discovery_port": 48899,
  "listen_repeats": 3,
  "state_flush_interval": 10000,
  "mqtt_state_rate_limit": 500,
  "mqtt_debounce_delay": 500,
  "mqtt_retain": true,
  "packet_repeat_throttle_sensitivity": 0,
  "packet_repeat_throttle_threshold": 200,
  "packet_repeat_minimum": 3,
  "enable_automatic_mode_switching": false,
  "led_mode_wifi_config": "Slow toggle",
  "led_mode_wifi_failed": "Fast toggle",
  "led_mode_operating": "Off",
  "led_mode_packet": "Flicker",
  "led_mode_packet_count": 1,
  "hostname": "milight-hub",
  "rf24_power_level": "MAX",
  "rf24_listen_channel": "HIGH",
  "wifi_static_ip": "",
  "wifi_static_ip_gateway": "",
  "wifi_static_ip_netmask": "",
  "packet_repeats_per_loop": 2,
  "home_assistant_discovery_prefix": "",
  "wifi_mode": "g",
  "default_transition_period": 500,
  "rf24_channels": [],
  "device_ids": [
    59016,
    59017
  ],
  "gateway_configs": [],
  "group_state_fields": [
    "state",
    "brightness",
    "mode",
    "color_temp",
    "bulb_mode",
    "computed_color"
  ],
  "group_id_aliases": {
    "Veranda": [
      "cct",
      59016,
      1
    ]
  }
}
```

Linkenelis, 2023-05-30:
Use a different device ID (0x001 or 1 should do it, unless you have more than 1 DIY units) and  you do need pairing.
Also make groups on this device (totally independent from the remote, as it also works without a remote)
If you have an MQTT somewhere connected, you will see both the remote and this device. Use software to keep them aligned? Off from remote => Off on device (like echo)

FilipDem, 2023-05-30:
I forgot to mention that I tried pairing with 0x689 and that it didn't work... Final goal is indeed using MQTT (connection to Domoticz), however want to get it work first with the UI/webserver. Normally there should not be a difference between 1 or other, but will give it a try this evening...
Anyway this is my only Mi Light device in house (we bought this house and it are integrated LEDs).

Linkenelis, 2023-05-30:
It should be working, if everything is correct.

- Receive is OK, so wiring 
Ahhhh => "rf24_channels": [],
Look at RF send channels in settings...
Mine is Low and High
[image]

My setup 7 groups, using device ID 1 and 2:
[image]
[image]

There is also has been a test device Device ID 3 and a wall switch (T4) with 0x2227.
See MQTT :
[image]

FilipDem, 2023-05-30:
Ok, thanks! Certainly a good point (the UI is not very clear on what is selected). Anyway played around with it and even copied your "repeat" settings that are higher than mine... All without positive result. Archhh...

Linkenelis, 2023-05-30:
Without positive result means pairing failed?

FilipDem, 2023-05-30:
Indeed... Become desperate...

Linkenelis, 2023-05-30:
Hmm, all that's left take my settings (you will need to change a group to CCT , WiFi and MQTT) running for several years now.

Save this as settings.json (including the {} ) and do a admin => restore with this file:
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.2.143:1883","mqtt_username":"name","mqtt_password":"password","mqtt_topic_pattern":"milight/cmnd/:device_id/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:group_id","mqtt_state_topic_pattern":"milight/state/:device_id/:group_id","mqtt_client_status_topic":"milight/info","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub_nodeMCU","rf24_power_level":"MAX","rf24_listen_channel":"MID","wifi_static_ip":"192.168.2.251","wifi_static_ip_gateway":"192.168.2.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","HIGH"],"device_ids":[1,2],"gateway_configs":[[1,5987,6],[2,5978,6]],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color","device_type"],"group_id_aliases":{"Achterkamer":["rgb_cct",2,4],"Eettafel":["rgb_cct",1,1],"Hal Lamp":["rgb_cct",2,2],"Hoeklamp":["rgb_cct",1,4],"Keuken Eiland":["rgb_cct",1,2],"Slaapkamer":["rgb_cct",2,3],"Test":["rgb_cct",2,1],"Zitkamer":["rgb_cct",1,3]}}

My about and settings
about:
{"firmware":"milight-hub","version":"1.10.6","ip_address":"192.168.2.251","reset_reason":"Software/System restart","variant":"d1_mini","free_heap":15536,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":26}}

settings:
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.2.143:1883","mqtt_username":"name","mqtt_password":"password","mqtt_topic_pattern":"milight/cmnd/:device_id/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:group_id","mqtt_state_topic_pattern":"milight/state/:device_id/:group_id","mqtt_client_status_topic":"milight/info","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub_nodeMCU","rf24_power_level":"MAX","rf24_listen_channel":"MID","wifi_static_ip":"192.168.2.251","wifi_static_ip_gateway":"192.168.2.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","HIGH"],"device_ids":[1,2],"gateway_configs":[[1,5987,6],[2,5978,6]],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color","device_type"],"group_id_aliases":{"Achterkamer":["rgb_cct",2,4],"Eettafel":["rgb_cct",1,1],"Hal Lamp":["rgb_cct",2,2],"Hoeklamp":["rgb_cct",1,4],"Keuken Eiland":["rgb_cct",1,2],"Slaapkamer":["rgb_cct",2,3],"Test":["rgb_cct",2,1],"Zitkamer":["rgb_cct",1,3]}}

FilipDem, 2023-06-03:
I google a bit to try to find the reference of the controller and as it is a single colour dimmer, I assume it is a FUT036. I cannot find this back in the list of tested device... Does anyone know if this works/should work?

Linkenelis, 2023-06-03:
FUT036 is a LED strip dimmer.
Single color is CCT. A single color bulb (CCT Bulb) should also work with RGB CCT setting. Sure color doesn't work and saturation works funny, but brightness, on/off yellow/white and pairing should work.

garmck, 2023-06-03:
> I google a bit to try to find the reference of the controller and as it is a single colour dimmer, I assume it is a FUT036. I cannot find this back in the list of tested device... Does anyone know if this works/should work?

I have a FUT036 and it works fine. You are right the FUT036 is a dimmer. 
Is the picture of the controller the same as yours? 
My label looks quite different to the one you have shown.

FilipDem, 2023-06-15:
The picture is indeed the one (I took it myself). So far no success... :-(.
Very strange at the Mi-Light hub can decode the remote control, I would suppose it sends the same signals. I fear I will need to test with a second setup (to verify if the Mi-Light hub is really sending), however have not a second NodeMCU... So will need to solder an esp12 (which is much more work).

garmck, 2023-06-15:
To be honest, the picture of your device looks like its a copy/fake - it looks different to the offical version.
[image]

Where did you buy this from ?

FilipDem, 2023-06-16:
I bought a house and got it with it :-). I think it is a rather old one, but rather sure it is a genuine one... The people who sold the house were elder people, and didn't do anything by theirselves but rather by professionals. It is part of the lighting of a nice veranda and the LED strips are integrated in the veranda itselves... So prety sure it is not somehow a fake product.
I assume I can stop hoping to get it working one day without buying a new one... Btw: this is my only device with MiLight for the rest I am using zigbee, 433Mhz, 868Mhz, Wifi devices...
Thanks for your thoughts anyway! And would there still be ideas, always welcome!

DigiH, 2023-07-21:
Just a quick thought, both the controller and the remote seem to be CCT only devices, but your screenshots show you setting it up as Remote Type **RGB+CCT.**

How is it working with Remote Type **CCT** only?