sidoh/esp8266_milight_hub #794: Compatibility Issues with Wifi 6?

mames13, 2023-06-12:
Hi! Quick question. 

Read the trouble shooting guide.  Esp definitely work but just won't connect to a new Access point (Netgear WAX206). 

Anyone know if there is a link to info on best practices for the Milight Hub and Wifi 6 settings ?

My Milight hub works great when using my old  2018 Wifi 5,  NETGEAR Nighthawk Extender as an access point. 

I have been trying to upgrade to a new Wifi 6 access point Netgear WAX206 (an inexpensive business-grade model with more controls) on what I think is a best possible replication of the 2.4GHz network from the old access point. 

I have tried lots of different configs but the hub just won't join. 

When I turn the old access 2018 Wifi 5 point  on the Milight Hub connects. 

Have tried resetting the wifi settings on the ESP via wifi manager as well and adding the new password, etc, with no luck. 

Any ideas would be great this has been torturing me for days.

Thanks!

mames13, 2023-07-06:
Has any one else had problems with using Wifi 6 vs Wifi 5?

Ninja1283, 2023-07-29:
Where does WiFi6 come into play at all? The ESP8266 and ESP32 based solutions all connect via WiFi4.

You just need to configure your WAX206 to broadcast a WiFi4 (2.4GHz) network, and connect the ESP to it.

sidoh, 2023-09-07:
Yep, WiFi 6 should be totally irrelevant.

I have noticed with my new router newer versions of the ESP8266 arduino SDK don't work unless I set wifi on the ESP to G mode but I'm still using an old version of the SDK for this project.

closed 2023-09-27 by sidoh