sidoh/esp8266_milight_hub #795: What release version is most compatible with NodeMCUv3?

mames13, 2023-07-06:
I'm building a new Milight-Hub with a NodeMCUv3 (31cm / larger board).

Which release best works with v3? 

I see the one that I have used successfully with the NodMCUv2 listed but I don't see one for v3.

Below are the release options  that I can currently see. 

Thank you. 

[esp8266_milight_hub_d1_mini-1.10.8.bin]
[esp8266_milight_hub_d1_mini_pro-1.10.8.bin]
[esp8266_milight_hub_esp07-1.10.8.bin]
[esp8266_milight_hub_esp12-1.10.8.bin]
[esp8266_milight_hub_huzzah-1.10.8.bin]
[esp8266_milight_hub_nodemcuv2-1.10.8.bin]

smarticz, 2023-09-26:
V2 working with V3
[esp8266_milight_hub_nodemcuv2-1.10.8.bin]

sidoh, 2023-09-27:
I think most of these binaries end up being the same thing. I'd probably cull them if I didn't think it'd wind up causing more confusion.

nodemcuv2 should work fine for v3.

closed 2023-09-27 by sidoh