sidoh/esp8266_milight_hub #803: Old backup in .json, new version want .bin

smarticz, 2023-09-26:
Hello, old versions create json backup and newest version want bin. It's possible to convert that file from json to bin?

Inside that files have different schema.

sidoh, 2023-09-26:
Yeah, the new format contains both aliases and a backup of the settings JSON. It's possible to convert but don't have a tool that does it.

You can still upload the .json file via `POST /settings` manually with `curl` or something similar and it'll have the same effect.

Are you having an issue with settings after upgrading?

smarticz, 2023-09-26:
Im downgraded to version with json support and uploaded backup. 
Now I updated to the latest version and it works :)

sidoh, 2023-09-26:
Gotcha. Glad you were able to figure it out.

Did you lose settings when you upgraded to 1.11?

smarticz, 2023-09-26:
Probably yes, because I upload backup and everything is fine.

Im not sure for 100%, but I have a lot of default settings, but Mqtt setup and automatic restart after 60 min were definitely taken from the backup.

sidoh, 2023-09-26:
You shouldn't need to restore a backup when upgrading -- they should all be transferred over seamlessly. There's a noisy warning in the firmware upgrade modal about backing up settings, but that's really a "just in case" kinda thing.

sidoh, 2023-09-27:
Gonna close this since it seems like we got it figured out, but let me know if I'm missing anything!

closed 2023-09-27 by sidoh