sidoh/esp8266_milight_hub #804: MQTT not working in HomeAssistant

R0SSI89, 2023-10-04:
I can't get the MQTT working with Home Assistant. I get the following error in the HomeAssistant logs. This happens when I add new alias in the milight interface.

Home Assistant log output:
```
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:201
Integration: MQTT (documentation, issues)
First occurred: October 2, 2023 at 23:29:03 (74 occurrences)
Last logged: October 3, 2023 at 07:16:45

Unable to parse origin information from discovery message, got {'support_url': 'http://10.10.10.213'}
```

### Setup information
HomeAssistant
```
Home Assistant 2023.9.3
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest
```

MQTT broker: Mosquitto broker (default settings, but created a user for it)
```
Current version: 6.3.1
```

#### Firmware version
1.11.0, didn't try another firmware

#### Output of http://milight-hub.local/about

```json
{
    "firmware": "milight-hub",
    "version": "1.11.0",
    "ip_address": "10.10.10.213",
    "reset_reason": "Power On",
    "variant": "nodemcuv2",
    "free_heap": 19584,
    "arduino_version": "3.1.2",
    "free_stack": 912,
    "flash_used": 10291,
    "flash_total": 957314,
    "flash_pct_free": 98,
    "queue_stats": {
        "length": 0,
        "dropped_packets": 0
    }
}
```

#### Output of http://milight-hub.local/settings
```json
{
    "admin_username": "--------------",
    "admin_password": "--------------",
    "ce_pin": 4,
    "csn_pin": 15,
    "reset_pin": 0,
    "led_pin": -2,
    "radio_interface_type": "nRF24",
    "packet_repeats": 50,
    "http_repeat_factor": 1,
    "auto_restart_period": 0,
    "mqtt_server": "--------------:1883",
    "mqtt_username": "--------------",
    "mqtt_password": "--------------",
    "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
    "mqtt_update_topic_pattern": "",
    "mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
    "mqtt_client_status_topic": "",
    "simple_mqtt_client_status": true,
    "discovery_port": 48899,
    "listen_repeats": 3,
    "state_flush_interval": 10000,
    "mqtt_state_rate_limit": 500,
    "mqtt_debounce_delay": 500,
    "mqtt_retain": true,
    "packet_repeat_throttle_sensitivity": 0,
    "packet_repeat_throttle_threshold": 200,
    "packet_repeat_minimum": 3,
    "enable_automatic_mode_switching": false,
    "led_mode_wifi_config": "Fast toggle",
    "led_mode_wifi_failed": "On",
    "led_mode_operating": "Slow blip",
    "led_mode_packet": "Flicker",
    "led_mode_packet_count": 3,
    "hostname": "milight-hub",
    "rf24_power_level": "MAX",
    "rf24_listen_channel": "HIGH",
    "wifi_static_ip": "",
    "wifi_static_ip_gateway": "",
    "wifi_static_ip_netmask": "",
    "packet_repeats_per_loop": 10,
    "home_assistant_discovery_prefix": "homeassistant",
    "wifi_mode": "n",
    "default_transition_period": 500,
    "rf24_channels": [
        "LOW",
        "MID",
        "HIGH"
    ],
    "device_ids": [
        4434
    ],
    "gateway_configs": [],
    "group_state_fields": [
        "state",
        "brightness",
        "color_temp",
        "bulb_mode",
        "computed_color",
        "effect",
        "device_id",
        "group_id",
        "color_mode"
    ]
}
```

### Additional context
First time using this and it would be great if I get it working (if I find some time I will provide some updates to the wiki)

bluebier, 2023-10-04:
i get the same error

sidoh, 2023-10-04:
Hey @R0SSI89 and @bluebier, which version of HomeAssistant are you guys using?  I'm guessing this field was added in recently.  I had hoped it'd add a link to the config page in the HA UI but it didn't, so probably safe to remove it.

KRiZ-R, 2023-10-04:
Hi @sidoh,
first time user here, and I have the same issue.
I am using the latest version of Home Assistant:

Home Assistant 2023.9.3
Supervisor 2023.09.2
Operating System 10.5
Frontend-versie: 20230911.0 - latest

R0SSI89, 2023-10-04:
@sidoh I added the HA version information in the issue, but it's the latest. I saw a lot of changes in the MQTT in HA, not sure if this could be the issue (I'm not a programmer, but a system engineer).

sidoh, 2023-10-04:
Thanks, guys. Will take a look. I'm also on 2023.9, so a little strange that I'm not seeing it.

R0SSI89, 2023-10-04:
Yeah you're correct, so I guess not a bug.. Maybe a misconfiguration, but where :(

sidoh, 2023-10-04:
Do you see an entry in `Settings > Devices` for espMH? It should be named the same thing as what you have the hostname configured to.

If you do, can you show me a screenshot of what yo usee there?

KRiZ-R, 2023-10-04:
Yes it is a warning in the logs.
It does not add the device under Settings > Devices

sidoh, 2023-10-04:
Gotcha. That's likely the cause. Will try to reproduce and get back to you guys.

R0SSI89, 2023-10-04:
For me it isn't either not showing under Settings > Devices.

R0SSI89, 2023-10-04:
Not suer if it helps, but these log lines appear in the Mosquito Broker logs when I remove a paired bulb and add it again (using the same device ID and group as the remote). 

```
2023-10-04 21:52:49: Client milight-hub-5582972 disconnected due to malformed packet.
2023-10-04 21:52:49: New connection from 10.10.10.213:50792 on port 1883.
2023-10-04 21:52:49: New client connected from 10.10.10.213:50792 as milight-hub-5582972 (p2, c1, k15, u'milight').
2023-10-04 21:53:02: Client milight-hub-5582972 disconnected due to malformed packet.
2023-10-04 21:53:02: New connection from 10.10.10.213:61083 on port 1883.
2023-10-04 21:53:02: New client connected from 10.10.10.213:61083 as milight-hub-5582972 (p2, c1, k15, u'milight').
2023-10-04 21:53:44: New connection from 172.30.32.2:57096 on port 1883.
2023-10-04 21:53:44: Client <unknown> closed its connection.
```

R0SSI89, 2023-10-04:
Ow and changed the MQTT state topic pattern to (compared to the settings provided when I created the issue):
```
milight/states/:hex_device_id/:device_type/:group_id
```

R0SSI89, 2023-10-04:
I got a notification about a new update: 2023.10.0. Should I try this update?

garmck, 2023-10-04:
> Do you see an entry in `Settings > Devices` for espMH? It should be named the same thing as what you have the hostname configured to.
> 
> If you do, can you show me a screenshot of what yo usee there?

This happened to me last weekend after I decided to start fresh, but hadnt got round to troubleshooting. (also shows IP adddress for URL)
@sidoh I suspect this is broken for new/fresh installs - I also suspect it wont add new aliases showing up for existing working installs.

garmck, 2023-10-04:
> I got a notification about a new update: 2023.10.0. Should I try this update?

I suspect it wont take @sidoh long to resolve this. But if you urgently need it working, you can manually add your lights using the MQQT Light component https://www.home-assistant.io/integrations/light.mqtt/#json-schema---examples

R0SSI89, 2023-10-04:
I'm not in a rush, I've got it "working" via the official Wifi box, but I want to make it a little smarter ;)

Thanks for the quick responses!

sidoh, 2023-10-04:
Taking a peek now. Thanks all for the extra info.

Don't expect a new version of HomeAssistant will help anything. I think I just mangled the discovery json a bit. (in my defense, the docs are kind of hard to parse).

garmck, 2023-10-04:
Here's how another integration shows in MQTT
[image]

This is the current MiLight MQTT
[image]

sidoh, 2023-10-04:
Yeah, `o` is supposed to be a supported abbreviation of `origin` according to [the docs](https://www.home-assistant.io/integrations/mqtt/). Maybe all of these fields need to be filled out? In any case, I couldn't really figure out what the function of this data was, so I just removed it.

I'd hoped it would add a link in the device dashboard to the config page, but it didn't have that effect.

sidoh, 2023-10-04:
I believe this guy will fix the problem: https://github.com/sidoh/esp8266_milight_hub/pull/805

I tagged a beta release. Will publish that when the build finishes for you guys to try out.

sidoh, 2023-10-04:
https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.11.1-beta1 give this guy a try

garmck, 2023-10-05:
Just loaded this version and can confirm the MQTT discovery is now adding devices as expected for myself.

R0SSI89, 2023-10-05:
Will the beta try tonight (I'm from the Netherlands). Thanks @sidon 😄

KRiZ-R, 2023-10-05:
Just did a quick test before getting ready to work, and it works!
It shows new devices immediately.
Thank you!

sidoh, 2023-10-05:
Nice, thanks for confirming, guys! I'll merge this and cut a release. LMK know if there are any issues :)

sidoh, 2023-10-05:
Fixed in [1.11.1](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.11.1)! Thanks everyone for the thorough reports.

R0SSI89, 2023-10-05:
Thank you, it works now!

closed 2023-10-05 by sidoh