sidoh/esp8266_milight_hub #806: brightness on display

KitKat-ok, 2023-10-08:
Im trying to display the current brightness on an oled display but I cant find any way to read it or any variable in the arduino code where can I find it or is there another way to achieve it ?

sidoh, 2023-10-08:
You're wanting to make some modifications to:

* get the current brightness of some arbitrary device/group combo
* display it on an external display

is that right?

you'd want to use `stateStore` defined in `main.cpp`. There are some example usages there that should be helpful, but it'd probably look something like this:

```c++
  BulbId myBulbId(0x1234, 1, MiLightRemoteType::REMOTE_TYPE_RGB_CCT);
  GroupState* state = stateStore->get(myBulbId);
  uint8_t brightness = state->getBrightness();
```

KitKat-ok, 2023-10-08:
Ill try if it works in a couple of days and I have another question how does it connect to wifi or does it create its own network ?

sidoh, 2023-10-08:
Check out the README: https://github.com/sidoh/esp8266_milight_hub#configure-wifi

`.local` is [mdns](https://en.wikipedia.org/wiki/Multicast_DNS#:~:text=In%20computer%20networking%2C%20the%20multicast,Domain%20Name%20System%20(DNS).). You don't need to use it. Can connect to it via hostname or IP address just as easily if your client doesn't support mdns.

KitKat-ok, 2023-10-08:
Can I completely disable mdns ?

sidoh, 2023-10-08:
The code is here! 

https://github.com/sidoh/esp8266_milight_hub/blob/master/src/main.cpp

Have fun :)

closed 2023-10-08 by sidoh