sidoh/esp8266_milight_hub #809: Editing the Interface

KitKat-ok, 2023-10-12:
How can I change the web interface changing the index.html doesnt seem to do anything

sidoh, 2023-10-12:
It needs to be compiled. There's a script that does it here, although you'll need to install node and npm:

https://github.com/sidoh/esp8266_milight_hub/blob/master/.build_web.py

ultimately everything winds up in this compiled progmem string, which is what's served from the web interface:

https://github.com/sidoh/esp8266_milight_hub/blob/master/dist/index.html.gz.h

going to close this since it was a question, but feel free to keep the discussion going if you have followups.

closed 2023-10-12 by sidoh

KitKat-ok, 2023-10-13:
I get an error when I try to execute it 
`Traceback (most recent call last):
  File "/esp8266_milight_hub/.build_web.py", line 8, in <module>
    Import("env")
    ^^^^^^
NameError: name 'Import' is not defined`

sidoh, 2023-10-13:
It's run as a part of the platformio build. I don't think you can run it directly. You'll need to compile and upload a firmware image with something like `platformio run -e d1_mini -t upload`. That should build the web image.

KitKat-ok, 2023-10-14:
Oh it works now thanks. I have another question how can I set the state of a bulb like turn it on or off in the arduino code ??

sidoh, 2023-10-14:
Here's an example: https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/WebServer/MiLightHttpServer.cpp#L363

KitKat-ok, 2023-10-17:
I cant get it to work can you explain it a bit further? @sidoh

sidoh, 2023-10-17:
The rough flow is something like this:

```c++
const uint16_t deviceId = 1234;
const uint8_t groupId = 1;
// config is a MiLightRemoteConfig. there are constants available. For example, FUT096Config is for rgbw
auto config = FUT096Config;

milightClient->prepare(&config, deviceId, groupId);
// call any methods on milightClient to change state:
milightClient->updateStatus(MiLightStatus::ON);
milightClient->updateBrightness(255);
```