sidoh/esp8266_milight_hub #810: Web interface not accessible

damiencorpataux, 2023-10-26:
One week setting up everything, the web server seems down:
- Ports `8266` and `6053` are open
    ```
    nmap 192.168.1.223 -p 1000-10000
    Starting Nmap 7.80 ( https://nmap.org ) at 2023-10-26 16:00 CEST
    Nmap scan report for milight (192.168.1.223)
    Host is up (0.0081s latency).
    Not shown: 8999 closed ports
    PORT     STATE SERVICE
    6053/tcp open  x11
    8266/tcp open  espeasy-p2p
    ```
  
- But port `80` is closed:
    ```
    nmap 192.168.1.223 -p 80
    Starting Nmap 7.80 ( https://nmap.org ) at 2023-10-26 16:01 CEST
    Nmap scan report for milight (192.168.1.223)
    Host is up (0.0095s latency).
    
    PORT   STATE  SERVICE
    80/tcp closed http
    ```

Am I missing something ?

I stripped the information below, because the problem is only about accessing the Web Interface.

### What firmware version(s) have you tried? 
I used
```
esphome:
  libraries:
    - milight=https://github.com/hencou/esphome-milight
```

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
NodeMCU

sidoh, 2023-10-26:
I'm not familiar with how @hencou has this set up. Perhaps he can help here or you can open up an issue over at https://github.com/hencou/esphome-milight ?

closed 2023-11-10 by sidoh

damiencorpataux, 2023-11-15:
True!
Many thanks for the accurate redirect ^.^