sidoh/esp8266_milight_hub #812: Posible NTP problems with internet restricted Wemos ESP8266

pvossel, 2023-11-01:
In my network I have alot of ESP8266 and ESP32 devices (Tasmota, WLED, Zehnder, etc). 
I want to restrict internet access for these devices. 
Only downside is, if they try and communicate with an external NTP server but cant reach it they become "unavailable" for a vew seconds. Something is happing to the WIFI. Not sure if its a reboot of the ESP8266 or reconnect with WIFI. 
I can mitigate this with most of my ESP8266 devices by using a local NTP server. 

Now I am also experiancing alot of "unavailable" Milight lights in Home Assistant. 
My assumption is that this is the same problem. When I give Milight Hub internet access I dont experiance the "unavailable" issue. 

Im running Milight Hub on a Wemos D1 mini (1.10.8 d1_mini).

Posible solution, configurable ntp server? 

HA Status: 
```
Milight Hub Milight2 turned on
21:05:53 - 14 minutes ago

Milight Hub Milight2 became unavailable
21:05:53 - 14 minutes ago

Milight Hub Milight2 turned on
20:50:53 - 29 minutes ago

Milight Hub Milight2 became unavailable
20:50:52 - 29 minutes ago

Milight Hub Milight2 turned on
20:35:52 - 44 minutes ago

Milight Hub Milight2 became unavailable
20:35:52 - 44 minutes ago

Milight Hub Milight2 turned on
20:20:52 - 1 hour ago

Milight Hub Milight2 became unavailable
20:20:52 - 1 hour ago
```

pvossel, 2023-11-02:
I've updated to the latest firmware 1.11.1 (d1_mini). 
Same issue here. 

Serial connection not showing any usefull logging for troubleshooting.

sidoh, 2023-11-26:
Hey guys, sorry for the late reply here.

I don't use NTP in this project. It's possible something else is using it behind the scenes, but I kind of doubt it.

I changed the default WiFi mode from N to G in 1.11 -- I suspect this might be the cause. Can you try swapping wifi mode to N?

pvossel, 2023-11-26:
It was already on N. 
Changed it to G. Problem solved!

closed 2023-11-26 by pvossel