sidoh/esp8266_milight_hub #814: Ability to rename saved controllers

SergioLuxx, 2023-11-09:
Good day!
Please add the ability to rename saved controller labels from the web interface. 
Now you can only completely remove the controller from the saved ones and pair it again with its new name

DigiH, 2023-11-09:
Just FYI, I have used the backup JSON before to rename labels, especially when the size limit was still an issue, and then uploaded the amended backup again with the renamings.

Not tried it yet with the new .bin format, but quickly looking at an opened backup it should still be possible.

SergioLuxx, 2023-11-10:
> Just FYI, I have used the backup JSON before to rename labels, especially when the size limit was still an issue, and then uploaded the amended backup again with the renamings.
> 
> Not tried it yet with the new .bin format, but quickly looking at an opened backup it should still be possible.

Unfortunately, it's not that simple. Perhaps with the English alphabet it actually works the way you suggested.
But using a non-English alphabet, the saved configuration files look like this:
Г§’1 3 РљРѕРјРЅР°С‚Р° fut089 1 1 {"admin_username":"","admin_password":"","ce_pin":4,... other settings...}
And it is not possible to edit Г§’1 3 РљРѕРјРЅР°С‚Р° fut089 1 1, much less understand what is written here.
And this is just one controller, for example, and not a full-fledged, working configuration, where there are many such incomprehensible names.

And besides, the developer will still be able to add the ability to rename directly from the web interface - this will be much easier to use than saving, editing, loading the full configuration

sidoh, 2023-11-10:
Yeah, I can look into adding support for updating the alias in the UI. 

For what it's worth, you can accomplish this now without editing the binary backup format by using [`PUT /aliases/:id`](https://sidoh.github.io/esp8266_milight_hub/branches/latest/#tag/Aliases/paths/~1aliases~1%7Bid%7D/put). You can get the ID by hitting `GET /aliases` which will list all aliases.

sidoh, 2023-11-13:
Give this a try. It's awkward and is just a wrapper around deleting the alias and allowing you to re-add it, but should hopefully be straightforward:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.11.2-beta1

The UI is kind of a mess so working with what's there.

SergioLuxx, 2023-11-15:
> Попробуйте это. Это неудобно и представляет собой всего лишь оболочку для удаления псевдонима и возможности его повторного добавления, но, надеюсь, все будет просто:
> 
> https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.11.2-beta1
> 
> Пользовательский интерфейс представляет собой беспорядок, поэтому работайте с тем, что есть.

[image]
Very good!
Yes, this actually works now and will help if you need to rename a specific controller!
Suggested improvements:
1) when you click on “rename”, display not an empty field, but the old name text.
2) have a "cancel" button.
Thank you!

sidoh, 2023-11-26:
Both good suggestions. I added (1), but (2) is a little too annoying to deal with. Some day I may rework the UI to be less of a clunky clusterfuck.

closed 2023-11-26 by sidoh