sidoh/esp8266_milight_hub #816: Error connecting to segmented network

SergioLuxx, 2023-11-19:
### Describe the bug

Good day.
I have two ESP8266 d1 mini controllers and two different installation locations.
In one place, the network is the usual 192.168.1.1; firmware 1.10.8 (which I used for a very long time) and the new 1.11.2-beta1 itself are connected to it without any problems. All you need to do is discover networks and enter a password.
Elsewhere, the network is segmented and has a special segment for smart devices 192.168.10.1; firmware 1.10.8 can be connected to it without any problems. All you need to do is discover networks and enter a password. But firmware 1.11... they simply refuse to connect.
They see the network, allow you to perform the initial configuration and after pressing the connect button, the controller freezes for about a minute, then reboots and on the configuration page it says that the connection failed.
I even tried to specify the network mask and gateway, change the Wi-Fi standards (b\g\n), but all to no avail.
Firmware 1.11... cannot connect to a segmented network.
I installed 1.10.8 back and connected without problems.

### Steps to reproduce

Install firmware 1.11.1 and try to connect to the segmented network

### Expected behavior

Successful connection and assignment of DHCP address

### Setup information

#### Firmware version
1.10.8 \ 1.11.1 \ 1.11.2-beta1

sidoh, 2023-11-19:
Hey @SergioLuxx, I'm guessing this is related to the upgrade of the Arduino SDK. I actually had some problems with my own network in n mode, but it worked fine after swapping to g. (I actually knew this was going to be an issue ahead of time because I encountered it with esphome a year or two ago).

Do you see anything interesting in serial logs?

SergioLuxx, 2023-11-19:
> Hey @SergioLuxx, I'm guessing this is related to the upgrade of the Arduino SDK. I actually had some problems with my own network in n mode, but it worked fine after swapping to g. (I actually knew [...]

Which magazines?
I flash the firmware using flash_download_tool_3.9.5 with complete memory clearing and downloading new firmware.
In all cases, the firmware is completed successfully, the board boots up, the indicator starts blinking quickly and an access point appears on the smartphone.
Connecting to it also goes without problems. Network Explorer finds all networks, including the smart devices segment.
All that remains is to set the password (but I tried the options and full network configuration)
That's all. A device with firmware 1.11.1 freezes and reboots after some time.
There is also no record in the router events that something tried to connect to it.
There are no other details. Although, again, I will mention that 1.10.8 worked and continues to work stably.

sidoh, 2023-11-19:
Yeah, I think I understand the issue. I'm just pretty sure that the problem is with the new version of the Arduino SDK. Unfortunately my options here were pretty limited -- I was either stuck with maintaining forks of very old versions of various libraries that added features or fixed bugs or upgrading to versions which are only compatible with newer versions of the Arduino SDK.

I'd suggest trying a minimal sketch like the following with the Arduino IDE and see if you can connect to the network you're having trouble with:

```c++
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.begin("<ssid>", "<password>");

  Serial.println();
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("success!");
  Serial.print("IP Address is: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // Wait a bit before scanning again
  delay(5000);
}
```

SergioLuxx, 2023-11-20:
> Yeah, I think I understand the issue. I'm just pretty sure that the problem is with the new version of the Arduino SDK. Unfortunately my options here were pretty limited -- I was either stuck with [...]

Okay, I'll check. I'll only be able to return to that apartment in about a week.

sidoh, 2023-11-24:
Thanks for testing it out! Can you tell me which version of the arduino SDK you're using? (`Ctrl + Shift + B` to open boards manager > search for "esp8266"):

[image]

After this, can you try a slightly more complicated sketch using the WiFiManager library?

Use `Ctrl + Shift + I` to open the library manager. Search for and install `WiFiManager`:

[image]

Then use this sketch:

```c++
#include <WiFiManager.h>
#include <strings_en.h>
#include <wm_consts_en.h>
#include <wm_strings_en.h>
#include <wm_strings_es.h>

void setup() {
  Serial.begin(115200);
  WiFiManager wifi;
  wifi.autoConnect("testwifi", "1234567890");
}

void loop() {
}
```

This is the same wifi setup flow the hub uses. You should see a network called `testwifi` with the password `1234567890`. See if this is able to connect.

SergioLuxx, 2023-11-24:
[image]
[image]
[image]

sidoh, 2023-11-24:
To be clear, should not change this line:

```c++
  wifi.autoConnect("testwifi", "1234567890");
```

This sketch will start a setup AP exactly the same as esp8266_milight_hub does. I want to test if you're able to connect to your AP with a minimal sketch using WiFiManager.

You'll want to connect to the AP it creates and go through the same setup process.

SergioLuxx, 2023-11-24:
[image]
Yes, everything worked out, I entered the data in the Wifi configurator

SergioLuxx, 2023-11-24:
[image]
I noticed that in this sketch there are no additional wifi configuration lines, such as static ip, mask, gateway, wifi stanbart.
These fields are available in firmware 1.11

sidoh, 2023-11-24:
Thanks for testing this out. A couple of things to further narrow down the issue:

1. Try this sketch which adds most of the customizations to wifimanager that espmh uses (you should hopefully not need to redo the setup process as wifi credentials should be saved):

```c++
#include <WiFiManager.h>
#include <strings_en.h>
#include <wm_consts_en.h>
#include <wm_strings_en.h>
#include <wm_strings_es.h>

WiFiManager wifi;

void setup() {
  Serial.begin(115200);
  WiFi.setPhyMode(WIFI_PHY_MODE_11G);

  wifi.autoConnect("testwifi", "1234567890");
  wifi.setConfigPortalBlocking(false);
  wifi.setConnectTimeout(20);
  wifi.setConnectRetries(5);

  wifi.setConfigPortalTimeout(180);
  wifi.setConfigPortalTimeoutCallback([]() {
      Serial.println(F("Wifi config portal timed out.  Restarting..."));
      delay(10000);
      ESP.restart();
  });
}

void loop() {
  wifi.process();
}
```

If that continues to work, paste the serial logs from esp8266_milight_hub after configuring wifi. May be helpful to see why wifimanager is failing to connect.

SergioLuxx, 2023-11-24:
[image]
There are no unusual logs. I downloaded a new sketch, rebooted the board and it immediately connected to the network

sidoh, 2023-11-24:
Thanks for testing! Can you try this?

> paste the serial logs from esp8266_milight_hub after configuring wifi. May be helpful to see why wifimanager is failing to connect.

Try flashing the same board with the latest version of espMH

SergioLuxx, 2023-11-24:
[image]
No, it doesn't work now

sidoh, 2023-11-24:
Thanks for testing. 

Can we try setting the phy mode to the default that espmh uses? Try this sketch:

```c++
#include <WiFiManager.h>
#include <strings_en.h>
#include <wm_consts_en.h>
#include <wm_strings_en.h>
#include <wm_strings_es.h>

WiFiManager wifi;

void setup() {
  Serial.begin(115200);

  WiFi.setPhyMode(WIFI_PHY_MODE_11B);

  wifi.setConfigPortalBlocking(false);
  wifi.setConnectTimeout(20);
  wifi.setConnectRetries(5);

  wifi.setConfigPortalTimeout(180);
  wifi.setConfigPortalTimeoutCallback([]() {
      Serial.println(F("Wifi config portal timed out.  Restarting..."));
      delay(10000);
      ESP.restart();
  });

  wifi.autoConnect("testwifi", "1234567890");
}

void loop() {
  wifi.process();
}
```

Apologies for the extended back and forth. I'm kind of having to take stabs in the dark here.

SergioLuxx, 2023-11-24:
[image]
This does not work.
In case it helps, I noticed that your individual sketches in the log output have a readable appearance at a connection speed of 115200. While the full milight firmware h
[image]
as a readable appearance at a speed of 9600

sidoh, 2023-11-24:
Yeah, baud rate shouldn't matter. 9600 is silly but I've not changed it to avoid confusion.

Can you try removing this line?

```c++
  WiFi.setPhyMode(WIFI_PHY_MODE_11B);
```

If it works without this, can you try with:

```c++
WiFi.setPhyMode(WIFI_PHY_MODE_11G);
```

and

```c++
WiFi.setPhyMode(WIFI_PHY_MODE_11N);
```

SergioLuxx, 2023-11-25:
> ```c++
> WiFi.setPhyMode(WIFI_PHY_MODE_11N);
> ```

[image]
This is after removing the line

[image]
WiFi.setPhyMode(WIFI_PHY_MODE_11G);

[image]
WiFi.setPhyMode(WIFI_PHY_MODE_11N);
Wifi unexpectedly connected successfully

sidoh, 2023-11-25:
Okay, so sounds like it does not work with B or G, but does with N. That's not surprising after the last exchange because the default is N.

I know you mentioned in your original post that you tried changing the mode during espMH setup, but can you try setting it to N again? 

I will verify that phy mode is actually getting swapped by that setting as well.

sidoh, 2023-11-25:
Okay, I think I figured out what's going on. Turns out the WiFiManager callback to save settings only gets called when it successfully connects to a network (lol).

So I think the summary of the problem here is:

1) Your network does not support B/G mode
2) espMH 1.11 uses G by default
3) WiFiManager was not saving your selection of wifi mode when you changed it, so you were stuck in G

I'll push a fix shortly that should allow you to actually set N mode, which should hopefully fix your problem.

sidoh, 2023-11-25:
Give this release a try: https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.11.2-beta3

You will need to go through the wifi setup process again. Make sure you set mode to N.

SergioLuxx, 2023-11-25:
> Give this release a try: https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.11.2-beta3
> 
> You will need to go through the wifi setup process again. Make sure you set mode to N.

Yes, it works!
I successfully updated, connected to the network and transferred all settings from firmware 1.10.8.
I also noticed a long connection to the network, I did not observe the port output, but it seems that first all connection attempts from 1 to 5 went through, then a reboot and after that a successful connection.
However, everything works!

sidoh, 2023-11-25:
Alright, great. I'll merge this and cut a non-beta release.

Thanks for your patience with the extended back-and-forth. Glad we figured it out!

closed 2023-11-26 by sidoh