sidoh/esp8266_milight_hub #818: Problem with fut089 remote detection

Bl00d-B0b, 2023-11-21:
Hello,

Looks like I found a problem with fut089. Maybe you can check source code, where it's an mistake:
I going to click on fut089 remote control groups, and show you raw packets:

**turn ON Group 1**
rgb_cct packet received (9 bytes):
Raw packet: 2D 85 39 86 45 24 2A 46 69 
Decoded:
Key      : 2D
b1       : 25
ID       : 3364
Command  : 01
Argument : 01
Sequence : 11
**Group    : 01**
Checksum : 15

**turno ON group 2:**
rgb_cct packet received (9 bytes):
Raw packet: 71 41 FD C2 09 E7 ED 09 EE 

Decoded:
Key      : 71
b1       : 25
ID       : 3364
Command  : 01
Argument : 02
Sequence : 12
**Group    : 02**
Checksum : DC

**turn OFF group 2:**
rgb_cct packet received (9 bytes):
Raw packet: DC FC CC 63 42 B3 85 3F 44 

Decoded:
Key      : DC
b1       : 25
ID       : 3364
Command  : 01
Argument : 0B
Sequence : 13
**Group    : 02**
Checksum : 71

**Turno OFF group 1:**
rgb_cct packet received (9 bytes):
Raw packet: 93 5C 27 6C 17 00 73 2E 85 

Decoded:
Key      : 93
b1       : 25
ID       : 3364
Command  : 01
Argument : 0A
Sequence : 14
**Group    : 02**
Checksum : 04

And now, when I try to press turn off any of groups, includin "0" it's always tryes to turn work with zone 2.

But If I enable zone 4 for example, then it's stucks on zone 4, does not care what zone I press.

Ed

sidoh, 2023-11-21:
I'll drag out my remote in a bit but I'm pretty sure this is just how the remote works. Definitely doesn't seem right to me, but those are just the packets its sending. I think for every command but "On," the group is the same of the last "On" you've pressed. 

Try doing this in this order:

Turn Group 1 ON 
Turn Group 2 ON
Turn Group 1 OFF (should turn G1 off)
Turn Group 2 ON (should have no effect since G2 is already on)
Turn Group 2 OFF (should turn G2 off)

Bl00d-B0b, 2023-11-23:
ok, then it's not a bug, sorry

closed 2023-11-26 by sidoh