sidoh/esp8266_milight_hub #825: MQTT Stops working

sincze, 2024-02-15:
### Describe the bug

I am running firmware": "milight-hub", "version": "1.11.2", on an EPS8266. I've recently switched to MQTT and that works excellent. Devices are automatically created using the Auto Discovery in Domoticz. I am running Mosquitto locally and MQTT explorer to monitor incoming messages.

For some reason the ESP decides not to send any more MQTT messages to Mosquitto. As a result lights can not be controlled anymore. To solve this. I need to login to the milight-hub , qo to MQTT settings page and press save.

[image]

### Steps to reproduce

Just wait and do nothing

### Expected behavior

If a connection to the MQTT server is lost for some reason I would expect the gateway to automatically retry this.
As I have done by going into the settings of mqtt and press save.

### Setup information

#### Firmware version
 "version": "1.11.2"

#### Output of http://milight-hub.local/about

```json
{
"firmware": "milight-hub",
"version": "1.11.2",
"ip_address": "192.1xxx.xxx.xxx",
"reset_reason": "Software/System restart",
"variant": "d1_mini",
"free_heap": 18064,
"arduino_version": "3.1.2",
"free_stack": 1488,
"flash_used": 10793,
"flash_total": 957314,
"flash_pct_free": 98,
"queue_stats": {
"length": 0,
"dropped_packets": 0
}```s"
```

#### Output of http://milight-hub.local/settings

```json
"{
"admin_username": "do_not",
"admin_password": "fool_me",
"ce_pin": 4,
"csn_pin": 15,
"reset_pin": 0,
"led_pin": -2,
"radio_interface_type": "nRF24",
"packet_repeats": 50,
"http_repeat_factor": 1,
"auto_restart_period": 0,
"mqtt_server": "mqtt.home.lan:1883",
"mqtt_username": "",
"mqtt_password": "",
"mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
"mqtt_update_topic_pattern": "milight/updates/:hex_device_id/:device_type/:group_id",
"mqtt_state_topic_pattern": "milight/states/:hex_device_id/:device_type/:group_id",
"mqtt_client_status_topic": "",
"simple_mqtt_client_status": false,
"discovery_port": 48899,
"listen_repeats": 3,
"state_flush_interval": 10000,
"mqtt_state_rate_limit": 500,
"mqtt_debounce_delay": 500,
"mqtt_retain": true,
"packet_repeat_throttle_sensitivity": 0,
"packet_repeat_throttle_threshold": 200,
"packet_repeat_minimum": 3,
"enable_automatic_mode_switching": false,
"led_mode_wifi_config": "Fast toggle",
"led_mode_wifi_failed": "On",
"led_mode_operating": "Slow blip",
"led_mode_packet": "Flicker",
"led_mode_packet_count": 3,
"hostname": "milight-hub",
"rf24_power_level": "MAX",
"rf24_listen_channel": "LOW",
"wifi_static_ip": "",
"wifi_static_ip_gateway": "",
"wifi_static_ip_netmask": "",
"packet_repeats_per_loop": 10,
"home_assistant_discovery_prefix": "milight",
"wifi_mode": "n",
"default_transition_period": 500,
"rf24_channels": [
"LOW",
"MID",
"HIGH"
],
"device_ids": [
2651,
51994
],
"gateway_configs": [
[
51994,
8899,
5
],
[
2651,
8898,
5
]
],
"group_state_fields": [
"state",
"brightness",
"mode",
"color_temp",
"bulb_mode",
"computed_color"
]
}```s"
```

### Additional context

sidoh, 2024-07-13:
Sorry for the late reply. There's definitely an attempt made to reconnect. This is called within `loop`:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MQTT/MqttClient.cpp#L125

```c++
void MqttClient::reconnect() {
  if (lastConnectAttempt > 0 && (millis() - lastConnectAttempt) < MQTT_CONNECTION_ATTEMPT_FREQUENCY) {
    return;
  }

  if (! mqttClient.connected()) {
    if (connect()) {
      subscribe();
      sendBirthMessage();

#ifdef MQTT_DEBUG
      Serial.println(F("MqttClient - Successfully connected to MQTT server"));
#endif
    } else {
      Serial.print(F("ERROR: Failed to connect to MQTT server rc="));
      Serial.println(mqttClient.state());
    }
  }

  lastConnectAttempt = millis();
}
```

I'd check the serial logs to see if there's any noise about MQTT disconnects.

sincze, 2024-07-14:
Tnx. I also locked the device to 1 AP to see if that helps. So it doesn't roam between multiple AP's.

As said it does obtain wifi but it only lost mqtt messaging traffic.