sidoh/esp8266_milight_hub #826: Home Assistant State Update Failing - "string indices must be integers, not 'str'"

mrrodge2020, 2024-02-22:
### What is the model number of the device you're trying to control?

FUT106

### What firmware version(s) have you tried? 

1.11.2 (nodemcuv2)

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 

Bought pre-built, not a hardware issue.

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 

RGB+CCT

### Have you tried controlling the device with a physical remote?

Works fine - don't know which remote I have.

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
{"firmware":"milight-hub","version":"1.11.2","ip_address":"75.0.1.110","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":17544,"arduino_version":"3.1.2","free_stack":768,"flash_used":14056,"flash_total":957314,"flash_pct_free":98,"queue_stats":{"length":0,"dropped_packets":0}}
```

#### /settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"10.0.0.4","mqtt_username":"user","mqtt_password":"12345","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[1,2,3],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type","oh_color","hex_color","color_mode"]}
```

### Problem Description

I'm using MQTT discovery in Home Assistant and the lights are discovered & can be controlled without issue from the HA interface but no colour, colour temperature or brightness states are updated when sending commands via HA, ESPMH web interface or remote.

My HA logs contain the following:

```
Exception in state_received when handling msg on 'milight/states/0x2/rgb_cct/1': '{"state":"ON","status":"ON","brightness":125,"level":49,"hue":248,"saturation":100,"color":"#2100FF","bulb_mode":"color","device_id":2,"group_id":1,"device_type":"rgb_cct","color_mode":"rgb"}' Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 43, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 483, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 364, in state_received self._update_color(values) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 230, in _update_color red = int(values["color"]["r"]) ~~~~~~~~~~~~~~~^^^^^ TypeError: string indices must be integers, not 'str' 
```

If I send a command to ESPMH via HA, the message shows up in the traffic sniffer, which is right as the light does as I've asked, but the UI of ESPMH doesn't update to reflect the selection either.

Selecting a mode in ESPMH does update the mode in HA, but nothing else seems to work.

Thanks!

mrrodge2020, 2024-02-22:
> You are using quite alot of Group States - you should only select the ones that are needed

I know... When it didn't work with the defaults I checked them all to see if it made a difference. No joy.

garmck, 2024-02-22:
Which defaults are you using?
no. 6 recommended as per - [https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant](url)

Do you get the same error?

mrrodge2020, 2024-02-23:
OK tried with the recommended 6 and it works better; no error in the logs, thanks!  I had to turn on the 'effect' group state to get the bulb mode state to work.  I now get a colour, brightness and effect state but no colour temperature.  I've enabled kelvin and color_temp and trie hue, with no luck.

sidoh, 2024-07-13:
Try these group state fields for homeassistant:

```
["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","computed_color","effect","device_id","group_id","device_type","color_mode"]
```

They're not all necessary but those are what I use and all modes work in my setup. The defaults should work with HomeAssistant as well. Going to close this but please feel free to reopen if there's still an issue

closed 2024-07-13 by sidoh