sidoh/esp8266_milight_hub #827: Home Assistant Integration uses depreciated flags

matthew-hz, 2024-04-04:
### Describe the bug

Home Assistant Integration uses depreciated flags.

Deprecated flags [rgb, color_temp] used in MQTT JSON light config for handling color mode, please use `supported_color_modes` instead. Got: {'device_class': 'light', 'schema': 'json', 'name': 'Back Columns 5', 'command_topic': 'milight/0x2002/rgb_cct/1', 'state_topic': 'milight/states/0x2002/rgb_cct/1', 'unique_id': '5FB8AA-Back Columns 5', 'device': {'name': 'milight-hub1', 'sw_version': 'esp8266_milight_hub v1.11.2', 'manufacturer': 'espressif', 'model': 'nodemcuv2', 'identifiers': '6273194', 'configuration_url': 'http://192.168.1.105'}, 'availability_topic': 'milight/client_status', 'payload_available': 'connected', 'payload_not_available': 'disconnected', 'brightness': True, 'effect': True, 'effect_list': ['night_mode', 'white_mode', '0', '1', '2', '3', '4', '5', '6', '7', '8'], 'rgb': True, 'color_temp': True, 'max_mireds': 370, 'min_mireds': 153, 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3

### Steps to reproduce

On startup of Home Assistant.

### Expected behavior

N/A

### Setup information

#### Firmware version
1.11.2

#### Output of http://milight-hub.local/about

```
{
    "firmware": "milight-hub",
    "version": "1.11.2",
    "ip_address": "192.168.1.105",
    "reset_reason": "Software/System restart",
    "variant": "nodemcuv2",
    "free_heap": 16600,
    "arduino_version": "3.1.2",
    "free_stack": 1600,
    "flash_used": 13805,
    "flash_total": 957314,
    "flash_pct_free": 98,
    "queue_stats": {
        "length": 0,
        "dropped_packets": 0
    }
}
```

#### Output of http://milight-hub.local/settings

```
{
    "admin_username": "",
    "admin_password": "",
    "ce_pin": 4,
    "csn_pin": 15,
    "reset_pin": 0,
    "led_pin": -2,
    "radio_interface_type": "nRF24",
    "packet_repeats": 50,
    "http_repeat_factor": 1,
    "auto_restart_period": 0,
    "mqtt_server": "192.168.0.10",
    "mqtt_username": "xxxxxxxx",
    "mqtt_password": "xxxxxxxx",
    "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
    "mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
    "mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
    "mqtt_client_status_topic": "milight/client_status",
    "simple_mqtt_client_status": true,
    "discovery_port": 48899,
    "listen_repeats": 3,
    "state_flush_interval": 10000,
    "mqtt_state_rate_limit": 500,
    "mqtt_debounce_delay": 500,
    "mqtt_retain": true,
    "packet_repeat_throttle_sensitivity": 0,
    "packet_repeat_throttle_threshold": 200,
    "packet_repeat_minimum": 3,
    "enable_automatic_mode_switching": false,
    "led_mode_wifi_config": "Fast toggle",
    "led_mode_wifi_failed": "Slow toggle",
    "led_mode_operating": "On",
    "led_mode_packet": "Flicker",
    "led_mode_packet_count": 3,
    "hostname": "milight-hub1",
    "rf24_power_level": "MAX",
    "rf24_listen_channel": "HIGH",
    "wifi_static_ip": "",
    "wifi_static_ip_gateway": "",
    "wifi_static_ip_netmask": "",
    "packet_repeats_per_loop": 10,
    "home_assistant_discovery_prefix": "homeassistant",
    "wifi_mode": "n",
    "default_transition_period": 500,
    "rf24_channels": [
        "LOW",
        "MID",
        "HIGH"
    ],
    "device_ids": [
        8193,
        8194,
        8195,
        8196,
        8200,
        8201
    ],
    "gateway_configs": [],
    "group_state_fields": [
        "state",
        "brightness",
        "mode",
        "color_temp",
        "bulb_mode",
        "computed_color",
        "device_id",
        "group_id",
        "device_type"
    ]
}
```

### Additional context

jkp1304, 2024-05-04:
Hi, i have the exact same problem.  Errors showing up i Home assistant. Anybody looking at the problem?

garmck, 2024-05-04:
This is not a MiLight bug. It is due to the changes in the MQTT Light configuration defined by Home Assistant.
The issue is explained in the first line:
**_"Deprecated flags [rgb, color_temp] used in MQTT JSON light config for handling color mode, please use supported_color_modes instead."_**

I would assume your YAML config has the following "rgb: true" or "color_temp: true"
You need to switch to the new config which is e.g. "supported_color_modes: ["rgb"]
[https://www.home-assistant.io/integrations/light.mqtt/#supported_color_modes](url)

sidoh, 2024-05-05:
left a comment on the related issue here: https://github.com/sidoh/esp8266_milight_hub/issues/823#issuecomment-2094858446

i won't be able to look at this for a while but happy to give some pointers if anyone is interested in giving it a shot.

sidoh, 2024-07-13:
Should be fixed in the [1.12.0-beta1](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.12.0-beta1) build. @matthew-hz can you try it out?

jkp1304, 2024-07-19:
Been using this beta release for a few days now. No errors at all in Homeassistant.  

Thank you :)

scottmckenzie, 2024-07-29:
Home Assistant is able to control my lights again after installing 1.12.0-beta1 build. Thanks!

Would it be possible to get file hashes added to the releases?

jellewie, 2024-09-16:
I also had the same error, I have now updated to 1.12.0-beta1 and we will see (So far so good)