sidoh/esp8266_milight_hub #828: Add support for FUT086 remote emulation

lyubomirtraykov, 2024-05-13:
Hello is it possible to add support for [FUT086](https://miboxer.com/product/8-zone-lora-433mhz-remote-control) remote emulation. It is using [SX1278](https://www.semtech.com/products/wireless-rf/lora-connect/sx1278) LoRa Transceiver. This will allow to control underwater lights.

sidoh, 2024-07-13:
Certainly sounds possible, but pretty far outside of the type of thing I'll have time to do myself. Happy to look at a PR if someone else has experience with that transceiver (and owns a light to test with).

PovilasID, 2024-07-14:
If you have an RTL-SDR usb dongle (HackRF or Noolec) and the remote you can just use rtl_433 to capture all unknown signals at your local lora frequency in the EU 868 Mhz and you should end up with .cu8 files that can be shared and either "replayed" or if the signal is undecrypted theoretically it should be decodable using tools like https://triq.org/pdv/