sidoh/esp8266_milight_hub #832: Update state via MQTT without sending a command to the lights.

solomondg1, 2024-07-11:
I have remotes, bulbs, and home assistant. If I configure it using the setup in the wiki where the ESP listens to the remote, publishes the state to MQTT, then a HASS automation sends a state back to the ESP via MQTT to finally control the light it takes about a second of delay for all that.

To avoid the delay I've also paired the remote with the light so it does the above and also instantly controls the light. A second later the light gets another command but that's usually fine for on/off.

The issue comes when you do a lot of changes and the lights go between what the remote is commanding and what the ESP is. I think the fix for this would be to have a way to tell the ESP there is a new state without it sending a command to the lights. I think this could either be a new MQTT topic or a variable in the JSON state payload.

Basically when the remote is controlling my lights all I want to do is keep home assistant's state in sync without sending packets to the lights from the ESP.

garmck, 2024-07-12:
Sounds like an over complicated setup.
You might want to paste some of your setup - including the MQTT settings you are using so we can see your config

Are you using the same device_id in home assistant and milight hub ?

sidoh, 2024-07-13:
I don't think you need the whole MQTT forwarder setup. Pair the bulb to the remote and use the same ID as the remote in the hub. As long as the ESP8266 receives the packet from the remote, its state will be updated accordingly.

solomondg1, 2024-07-14:
@sidoh I've added the device ID of the remote to the ESP and I'm able to control the light with that device ID. If I turn the light off via the remote the state on the ESP remains on. I tried refreshing the page and confirming the ESP is receiving data from the remote via the sniffing log.