sidoh/esp8266_milight_hub #836: How to flash ESP8266 with PlatformIO

NC747GitHub, 2024-08-17:
Hi everyone,

I do apologize to ask you that but I don't manage to flash my ESP8266 with PlatformIO...
Can someone please help me to do it?
I spent few days trying to do it by reading forums and watching videos but I'm completly lost.
I want to flash my Wemos D1 mini v4 with PlatformIO in oder to have the milight program on it, it says it's easy but unfortunatly not for me.

Thanks for any help...

xury77, 2024-08-23:
You don't need PlatformIO to flash your device. Just download latest binary for your board and flash it using for eq Tasmotizer or any other flashing software (esptool etc)
https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.11.2

DonnyVerduijn, 2024-08-26:
I was facing some unexpected behavior in vscode using the PlatformIO dev container. When i tried uploading from master directly, it didn't work. Resetting to an earlier commit and restoring the changes after that commit in the working directory made it suddenly work (at least without any other changes to the repository), could be unrelated to the repository though.