sidoh/esp8266_milight_hub #838: Deprecated option used in mqtt publish action call

roydestroy, 2024-09-22:
### Describe the bug

Home assistant is producing the following message.

Deprecated payload_template option used in MQTT publish action call to topic milight/commands/0xC9A6/fut091/1 from payload template {"command":"brightness_up"}. Use the payload option instead. In automations templates are supported natively. Update the automation or script to use the payload option instead and restart Home Assistant to fix this issue.

### Steps to reproduce

### Expected behavior

### Setup information

#### Firmware version
1.11.2
1.12.0-beta1

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.12.0-beta1","ip_address":"192.168.1.27","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":18592,"arduino_version":"3.1.2","free_stack":1072,"flash_used":18574,"flash_total":957314,"flash_pct_free":98,"queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.1.42:1883","mqtt_username":"mqttuser","mqtt_password":"","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"Slow blip","led_mode_operating":"On","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"HIGH","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"g","default_transition_period":500,"rf24_channels":["HIGH"],"device_ids":[51622],"gateway_configs":[],"group_state_fields":["state","brightness","mode"]}
```

### Additional context
[image]

sidoh, 2024-09-27:
How are you integrating with homeassistant?

sidoh, 2024-09-27:
Sorry, I meant how is espMH integrated with homeassistant. I'm guessing you're using mqtt discovery and you haven't manually configured a light entity.

This warning seems to indicate that:

a) homeassistant itself is sending this command on the mqtt commands topic
b) something other than homeassistant is sending commands on the mqtt commands topic

If I'm reading the error correctly it kind of sounds like an automation you have set up is the source? This looks related: https://community.home-assistant.io/t/deprecated-payload-template-option-used-in-mqtt-warning/765187

roydestroy, 2024-09-27:
Well, I have this automation so that I can forward the commands from the physical controller to homeassistant. I don't know if this is the culprit and if it is related to it.

```
alias: MiLight Forwarder
description: ""
trigger:
  - platform: mqtt
    topic: milight/updates/0xC9A5/fut091/+
  - platform: mqtt
    topic: milight/updates/0x45F/cct/+
condition: []
action:
  - metadata: {}
    data:
      topic: milight/commands/0xC9A6/fut091/1
      payload_template: "{{ trigger.payload }}"
    action: mqtt.publish
mode: single

```

sidoh, 2024-09-27:
Yep, that looks like it's the culprit. I'm not really familiar with the syntax but the warning suggests you can just change `payload_template` to `payload`.

sidoh, 2024-09-29:
gonna close, but let me know if something got missed :)

closed 2024-09-29 by sidoh