sidoh/esp8266_milight_hub #839: Any way to set multiple lights in a single http request?

furahivszuri, 2024-10-15:
Hello,

I control 15 Milight bulbs set up around a room. Some are RGBW, some RGB_CCT. I wrote a page that allows me to program them. Basically a simple JSON config lets you set all bulbs to the same color, each to a different color or even animations (with either all bulbs animating at the same time, or each bulb individually). I don't sync to music or anything crazy, so precise timings are not important, but quick changes are

I originally ran it with Milight controllers, and for the most part it worked

I recently moved to an esp8266 controller with ESPMH, and I've noticed that it's relatively easy to send too many http requests in a short time to the hub. The web server stops responding after a while. 

I've tried many tings, like sending or not sending blockOnQueue, awaiting each request (which I am currently doing fro a browser), that ends up being too slow; sending a few requests staggered and awaiting them, etc...

The thing is for many of these animations, I'll really have 3 or 4 different colors, so if i could send a request like "PUT gateways/1/rgbw/1, gateways/1/rgbw/2, gateways/1/rgb_cct/1" with a body of {"hue":240,"saturation":60,"level":100,"status":"on"} , tht would save me a lot of requests

Even better if a single request could set all the lights at once (like set 1 and 2 to yellow, 3 and 4 to blue, 5 and 6 to red). I get there is only one radio, so they will actually execute sequentially, that's fine, a single request for all the lights is easier to await compared to one per light, the overhead adds up quickly

Failing that, any recommendations on how to tune the software and my requests for best performance? 

Thanks

sidoh, 2024-10-15:
There's not currently a way to do this with the API, but I'm open to adding one! There was at some point a way to fan out eg:

```
PUT /gateways/1,2,3/rgbw,rgb_cct/1,2
```

but that would send the cartesian product of all of those (so 3*2*2 in this case). It's possible that still works actually, but looking at the code I think I broke that ages ago during a migration. If it's useful in that form it's a relatively simple fix (these `TokenIterator`s [here](https://github.com/sidoh/esp8266_milight_hub/blame/master/lib/WebServer/MiLightHttpServer.cpp#L503) need to have the third argument set to `,`) .

Think I'd rather add a separate endpoint like `PUT /gateways` or `PUT /gateways/batch` or something like that which accepts an array of IDs in the body as you're suggesting. Also not very hard to add that! `lib/WebServer` is the place to look if you're interested in taking a crack at it.

furahivszuri, 2024-10-15:
Thanks! I'll have a look and give it a try myself!

If I don't stop myself though, I might end up trying to add the whole animation engine (maybe in a fork, I'm not sure that code would be super useful to everyone else)

sidoh, 2024-10-21:
Hey @furahivszuri, I was already in the code updating the web UI and had a place where I wanted a batch update endpoint anyway. It's added in [1.13.0-beta2](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.13.0-beta2).

Documentation is [https://sidoh.github.io/esp8266_milight_hub/branches/1.13.0-beta2/#tag/Device-Control/paths/~1gateways/put](here).

Syntax is something roughly like:

```
curl -v -X PUT -d '[{"gateways":[{"device_id":4493,"device_type":"rgb_cct","group_id":1}],"update":{"state":"ON"}}]' 10.133.8.109/gateways
```

There are a few levels of nesting to make it more generic -- hopefully not too much of a pain in the butt to use!