sidoh/esp8266_milight_hub #841: unable to open settings / http://192.168.0.197/#/settings keeps loading

angelgc95, 2024-10-25:
[image]

### What is the model number of the device you're trying to control?

### What firmware version(s) have you tried? 

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 

### Have you tried controlling the device with a physical remote?

 

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
"... /about output. put between the ```s"
```

#### /settings

```json
"... /settings output. put between the ```s"
```

sidoh, 2024-10-25:
Can you take a screenshot of your developer console with the network tab open?

angelgc95, 2024-10-25:
[image]

sidoh, 2024-10-25:
Can you reload the page while it's open?

Also try loading `http://192.168.0.1/settings` and `http://192.168.0.1/about`

angelgc95, 2024-10-25:
> Can you reload the page while it's open?
> 
> Also try loading `http://192.168.0.1/settings` and `http://192.168.0.1/about`

-  `http://192.168.0.1/settings` and `http://192.168.0.1/about` shows: No such file or directory

-

sidoh, 2024-10-25:
errr, sorry, `http://192.168.0.197/settings` etc

And if you could also take a screenshot of the dev console after reloading the page

angelgc95, 2024-10-25:
[image]

> errr, sorry, `http://192.168.0.197/settings` etc
> 
> And if you could also take a screenshot of the dev console after reloading the page

angelgc95, 2024-10-25:
http://192.168.0.197/about

{"firmware":"milight-hub","version":"1.13.0-beta1","ip_address":"192.168.0.197","reset_reason":"External System","variant":"nodemcuv2","free_heap":19080,"arduino_version":"3.1.2","free_stack":1200,"flash_used":983090,"flash_total":0,"flash_pct_free":0,"queue_stats":{"length":0,"dropped_packets":0}}

I did had the previous version (and was working good) but i had to wipe the esp to flash the new firmware

sidoh, 2024-10-25:
Oh no, that's a drag! Sorry you're running into trouble.

Yeah, it looks like your settings file is empty (if you could try loading `http://192.168.0.197/settings` manually just to make sure that'd be great). That's probably because there's something funky with the flash configuration. Hopefully we can figure out what's going on.

Did you have to erase flash because you had issues after upgrading?

Could you check serial logs?

angelgc95, 2024-10-25:
[No is ju=](http://192.168.0.197/settings) empty 

No, I just couldn't figure out how to update wifi settings

logs: 
--- Miniterm on /dev/cu.usbserial-0001  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�␄l␌��␌␌␌c␌o�|␂�␄��␄�p␄�|~�␀l��␀$`␂�␒␓on␄l`␂␎␂n{���n␌␄#$�␏lp�n�␐␃␄␌r��ܜ␜��␃␄b␄n�|␃␄��␌b��on�␂␄␌$`␂�␛␒no␌$`␃␎␃or����␂␌␌c��`␂␎r��N␌␌B�p{l�␂␌␄��␌��$`␂��o�␎l��␂�␒�N�{��N|�␄�$��l`␂␜B�2␃␄�|␂s��d�o�␌�n�␀␌$`␃��r�d�␜�␛�␌␌␌�␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀

sidoh, 2024-10-25:
Looks like you have the baudrate set to 115200. Try setting it to 9600.

angelgc95, 2024-10-25:
--- Miniterm on /dev/cu.usbserial-0001  9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
b␄�␘I␁�␔C�bE�Failed to mount file system
Opening settings file failed
*wm:AutoConnect 
*wm:Connect Wifi, ATTEMPT # 1 of 5
*wm:Connecting to SAVED AP: IoT_8
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.0.197
Wifi connected succesfully

Setup complete (version 1.13.0-beta1)
Opening settings file failed
Opening settings file failed
Unhandled websocket event: 10

sidoh, 2024-10-25:
Okay yeah that confirms there's something going on with flash:

```
(...)
Failed to mount file system
(...)
```

I would try the following:

* Wipe flash entirely (I do this with a command like `python3 -m esptool -b 115200 erase_flash`)
* Reflash the firmware (`python -m esptool -b 460800 write_flash 0x0 <firmware_file>`)

Let me know what you see

thoness-HA, 2025-01-06:
Hi, I have the a similar problem with the webpage interface: 

[image]

[image]

[image]

My settings gives the follwing output:
`{"admin_username":"admin","admin_password":"Password4711","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":1440,"mqtt_server":"192.168.22.203:1883","mqtt_username":"mqtt-user","mqtt_password":"Password4711","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/state/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"LED","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.168.22.204","wifi_static_ip_gateway":"192.168.22.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant/","wifi_mode":"g","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[],"gateway_configs":[],"group_state_fields":["state","brightness","computed_color","mode","color_temp","color_mode"]}`

the about gives the following
`{"firmware":"milight-hub","version":"1.13.0","ip_address":"192.168.22.204","reset_reason":"External System","variant":"esp12","free_heap":13080,"arduino_version":"3.1.2","free_stack":1168,"flash_used":8534,"flash_total":957314,"flash_pct_free":99,"mqtt":{"configured":true,"connected":true,"status":"Connected"},"queue_stats":{"length":0,"dropped_packets":0}}`

I was using the firmware 1.13.0 on the device, it happened twice on different ESPs, I'm using nodemcu lua v3 from Ali, just using FUT103

It did work perfectly. I was able to set up 20 bulbs, up to one day, the webpage is not accessible anymore. 

and more interesting: my HomeAssistant is still able to adresse the bulbs via the ESP through MQTT, it just seems the webpage down. tried on firefox, chrome and chrome incognito.

and the switch between dark and bright **does** work

last time I just flashed it again, but the problem reoccurred now. 

Any ideas on that?

sidoh, 2025-01-06:
Ack, that's annoying! Sorry you're running into issues.

I'm guessing there's some kind of UI error that isn't being surfaced. Can you include a screenshot of your dev console?

thoness-HA, 2025-01-07:
But the use of the software is FANTASTIC btw.... Thanks a lot

here is a video of using it with the open console

[https://youtu.be/6GhWM_SSM6I](url)

what else do you need?

sidoh, 2025-01-07:
Perfect, thank you. Can you share the output of `/gateways`?

thoness-HA, 2025-01-07:
`[{"device":{"alias":"0x1001 Licht Flur Haustür oben ","id":1,"device_id":4097,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x1002 Licht Flur Haustür oben ","id":2,"device_id":4098,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x1003 Licht Flur Haustür unten","id":3,"device_id":4099,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x1004 Licht Flur Haustür unten","id":4,"device_id":4100,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x1010 Licht Flur Haustür Grupp","id":17,"device_id":4112,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x2001 Licht Haustür rechts","id":7,"device_id":8193,"group_id":1,"device_type":"rgb_cct"},"state":{"state":"OFF","color_mode":"color_temp","level":100,"kelvin":100}},{"device":{"alias":"0x2002 Licht Haustür links","id":6,"device_id":8194,"group_id":1,"device_type":"rgb_cct"},"state":{"state":"OFF","color_mode":"color_temp","level":20,"kelvin":100}},{"device":{"alias":"0x2010 Licht Haustür Gruppe","id":18,"device_id":8208,"group_id":1,"device_type":"rgb_cct"},"state":{"state":"ON","color_mode":"color_temp","level":10,"kelvin":100}},{"device":{"alias":"0x3001 Licht Flur Hof oben recht","id":11,"device_id":12289,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x3002 Licht Flur Hof oben links","id":12,"device_id":12290,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x3003 Licht Flur Hof unten rech","id":13,"device_id":12291,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x3004 Licht Flur Hof unten link","id":14,"device_id":12292,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x3010 Licht Flur Hof Gruppe","id":15,"device_id":12304,"group_id":1,"device_type":"rgb_cct"},"state":{"state":"ON","color_mode":"color_temp","level":10,"kelvin":100}},{"device":{"alias":"0x4001 Licht Hof Mistbunker link","id":19,"device_id":16385,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x4002 Licht Hof Mistbunker mitt","id":20,"device_id":16386,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x4003 Licht Hof Mistbunker rech","id":21,"device_id":16387,"group_id":1,"device_type":"unknown"},"state":{}},{"device":{"alias":"0x4004 Licht Hof Scheune","id":23,"device_id":16388,"group_id":1,"device_type":"rgb_cct"},"state":{"state":"OFF","color_mode":"color_temp","level":20,"kelvin":100}},{"device":{"alias":"0x4010 Licht Hof Gruppe","id":22,"device_id":16400,"group_id":1,"device_type":"rgb_cct"},"state":{"state":"ON","color_mode":"color_temp","level":10,"kelvin":100}}]`

Golgotha84, 2025-02-11:
Hello, got the exact same issue. I'm undating to 1.13.1 beta 2, just in case it changes something. Keep you updated.

Golgotha84, 2025-02-11:
And, just a big thanks for this solution which is so good for those like us who are using miboxer bulbs ! AMazing work from all aspects, even security one!

Cliff-R-K, 2025-12-04:
Any Progress?