sidoh/esp8266_milight_hub #842: Can't program/Compile on ESP32

s0k33t, 2024-10-29:
### What firmware version(s) have you tried? 

1.13.0-rc1
1.13.0-beta2
1.13.0-beta1
1.12.0-beta1

### Which ESP8266 board are you using?

ESP32 WROM-32

Can't program it. 

With the precompiled version I get:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x2068746f

And whit Platformio, can't compile it.
.pio\libdeps\esp32\WebSockets\src\WebSocketsClient.cpp:250:30: error: 'class WiFiClientSecure' has no member named 'setCACertBundle'
                 _client.ssl->setCACertBundle(_CA_bundle);

sidoh, 2024-10-29:
Hmm not being able to compile is a little surprising. Have compiled it locally recently and [there's a CI build](https://github.com/sidoh/esp8266_milight_hub/actions/runs/11534861664/job/32109340139) which is working. I also just tried deleting my build directory and recompiling and it works.

Checking `platformio.ini`, I bet the issue is I'm not pinning a version of the platform and you have an older version cached. Can you try updating the `[esp32]` configuration to look like this?

```ini
[esp32]
extends = base
platform = espressif32@~6
(...)
```

As for flashing the pre-compiled binary, what command are you using to do that?

sidoh, 2024-10-29:
Pinned the version in this open PR: https://github.com/sidoh/esp8266_milight_hub/pull/840/commits/4a117a7672014fedc86ffbc25ad3f7aae7e2dfa1

sidoh, 2024-10-30:
Glad to hear it! 

Yeah, I think for esp32 the address is `0x1000`. There's some system stuff like a partition table (maybe it's just that?) at the first area of flash.

closed 2024-10-30 by sidoh