sidoh/esp8266_milight_hub #844: Remote fut092 recognized as fut089

Jojo-1000, 2024-11-14:
### Describe the bug

First of all, thank you for this great project. It was exactly what I was looking for. I am trying to control a fut035w+ cct led controller with a fut092 remote, which works alright.
The sniffer shows the remote messages (after switching to the mid listening channel) but the type and some commands, such as color temperature, are not decoded properly. They just show up as "button_id: 3". Am I doing something wrong or is there no way to distinguish between these remote types? If that is not possible, I would like some sort of override setting to manually set the type.

If you can point me in a rough direction, I am willing to implement and try some code changes myself.

### Steps to reproduce

- Rf24 Listen Channel: MID in settings
- Press "on" on the remote
- Press on the color temperature select
- Press on the saturation select
- See captured events below

### Expected behavior

- Device type should show as fut092, if possible
- button_id 3 should be decoded as color temperature (0 warmest, 100 coolest)
- button_id 4 should be decoded as saturation (0 to 100)
- The state/color mode should be updated accordingly

All other buttons seem to work correctly.

### Setup information

#### Firmware version

1.13.0-2-g0a9ab66

#### Output of http://milight-hub.local/about
<details>

```json
{
  "firmware": "milight-hub",
  "version": "1.13.0-2-g0a9ab66",
  "ip_address": "192.168.178.150",
  "reset_reason": "External System",
  "variant": "d1_mini",
  "free_heap": 16096,
  "arduino_version": "3.1.2",
  "free_stack": 1168,
  "flash_used": 3765,
  "flash_total": 957314,
  "flash_pct_free": 99,
  "mqtt": {
    "configured": true,
    "connected": true,
    "status": "Connected"
  },
  "queue_stats": {
    "length": 0,
    "dropped_packets": 0
  }
}
```
</details>

#### Output of http://milight-hub.local/settings

<details>

```json
{
  "admin_username": "",
  "admin_password": "",
  "ce_pin": 4,
  "csn_pin": 15,
  "reset_pin": 0,
  "led_pin": -2,
  "radio_interface_type": "nRF24",
  "packet_repeats": 50,
  "http_repeat_factor": 1,
  "auto_restart_period": 0,
  "mqtt_server": "homeassistant:1883",
  "mqtt_username": "mqtt-milight",
  "mqtt_password": "***",
  "mqtt_topic_pattern": "milight/commands/:device_id/:device_type/:group_id",
  "mqtt_update_topic_pattern": "",
  "mqtt_state_topic_pattern": "milight/state/:device_id/:device_type/:group_id",
  "mqtt_client_status_topic": "milight/client_status",
  "simple_mqtt_client_status": true,
  "discovery_port": 48899,
  "listen_repeats": 3,
  "state_flush_interval": 20000,
  "mqtt_state_rate_limit": 500,
  "mqtt_debounce_delay": 500,
  "mqtt_retain": true,
  "packet_repeat_throttle_sensitivity": 0,
  "packet_repeat_throttle_threshold": 200,
  "packet_repeat_minimum": 3,
  "enable_automatic_mode_switching": false,
  "led_mode_wifi_config": "Fast toggle",
  "led_mode_wifi_failed": "On",
  "led_mode_operating": "Slow blip",
  "led_mode_packet": "Flicker",
  "led_mode_packet_count": 3,
  "hostname": "milight-hub",
  "rf24_power_level": "HIGH",
  "rf24_listen_channel": "MID",
  "wifi_static_ip": "",
  "wifi_static_ip_gateway": "",
  "wifi_static_ip_netmask": "",
  "packet_repeats_per_loop": 10,
  "home_assistant_discovery_prefix": "homeassistant/",
  "wifi_mode": "g",
  "default_transition_period": 500,
  "rf24_channels": [
    "MID"
  ],
  "device_ids": [],
  "gateway_configs": [],
  "group_state_fields": [
    "state",
    "brightness",
    "computed_color",
    "mode",
    "color_temp",
    "color_mode"
  ]
}
```
</details>

### Additional context
**Captured events:**
On press:
```
Device ID: 44919
Group ID: 0
Remote Type: fut089
Packet: 02 A5 FE 4D D2 A4 00 91 C1

State:
{
  "state": "ON",
  "color": {
    "r": 255,
    "g": 0,
    "b": 157
  }
}
Command:
state: "ON"
```
Color temperature:
```
Device ID: 44919
Group ID: 0
Remote Type: fut089
Packet: 56 51 9A F9 38 43 95 F5 2F

State:
{
  "state": "ON",
  "color": {
    "r": 255,
    "g": 0,
    "b": 157
  }
}
Command:
button_id: 3
argument: 75
```
Saturation:
```
Device ID: 44919
Group ID: 0
Remote Type: fut089
Packet: F2 B5 0E 7D 9D 9B 2B 61 FE

State:
{
  "state": "ON",
  "color": {
    "r": 255,
    "g": 0,
    "b": 157
  }
}
Command:
button_id: 4
argument: 47
```

sidoh, 2024-11-15:
Hello! The entry point for packet decoding is here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/src/main.cpp#L163

All it's doing is:

* Looping through each remote type it knows about
* Filtering down to remotes that use the same radio configs as the received packet
* Asks the handler if it can handle the given packet (in most cases, this is just checking packet length and something like a protocol ID)

The `canHandle` method for the fut089 is [here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/V2PacketFormatter.cpp#L13).

It's possible milight released a different remote which uses the same protocol ID has fut089 (an 8-group RGB+CCT remote). fut092 looks exactly like fut091.

Happy to provide pointers or help out however I can, but will be hard for me to be useful beyond that without a physical device to test with.

Jojo-1000, 2024-11-15:
> It's possible milight released a different remote which uses the same protocol ID has fut089 (an 8-group RGB+CCT remote). fut092 looks exactly like fut091.

This is probably true. I looked at the manual it came with and it had one table for fut88,89,89-B,89S,92 and 92-B. They probably use the same protocol just with different buttons.
As far as I can tell, there are different protocol IDs for the other V2 packets, but these are both using 0x25. So there is no way to add a different class for it.

My idea is to add the missing commands here:

https://github.com/sidoh/esp8266_milight_hub/blob/0a9ab66566f06e33352811a8f71b47cf3f1bdcf1/lib/MiLight/FUT089PacketFormatter.h#L8-L16

Maybe I will rename the current saturation to `kelvin_sat_mixed` to avoid confusion.

And then insert the decoding like the current saturation, except it is not determined by mode:
https://github.com/sidoh/esp8266_milight_hub/blob/0a9ab66566f06e33352811a8f71b47cf3f1bdcf1/lib/MiLight/FUT089PacketFormatter.cpp#L140-L147

If all lights support these separate color temperature and saturation commands, it would also allow to change the packet encoding and make it simpler. Then it wouldn't fail unexpectedly when the internal color mode is wrong.

Jojo-1000, 2024-11-15:
I added the command codes and it now shows the correct commands. Switching the state from rgb to color_temp also works, but I assume it should also switch back to rgb when using the saturation command.

So I have:
- Press hue: `color_mode: rgb`
- Press color temperature: `color_mode: color_temp`
- Press saturation: `color_mode: color_temp`
- Press hue: `color_mode: rgb`, but the correct saturation is applied in the state

I can't test this with my non-rgb light, but do you have any experience with how it should behave?

PR #846 (2024-11-23): Add support to decode the FUT092 remote
It uses the same protocol as FUT089, but has separate commands for color temperature and saturation.

Closes #844

Please review my naming changes (I used FUT092 in the same enum as FUT089 options, not sure if that is appropriate).

I chose not to change the `updateTemperature` and `updateSaturation` methods, because that would require testing on a RGB light. However, it should be possible to use the new commands to avoid the current mode-dependent behavior.
closed 2025-02-15 by Jojo-1000