sidoh/esp8266_milight_hub #851: changing modes

KitKat-ok, 2024-11-23:
I have updated the firmware to the newest version and I noticed that some buttons disappeared from the rgb remote type like the buttons for setting modes and the speed why were these buttons deleted ? and could they be added back I have attached a picture of what buttons I mean and a picture of the rgb remote view  
[image]
[image]

sidoh, 2024-11-24:
Added these buttons back in 1.13.1-beta1

dantesia, 2024-12-05:
> Added these buttons back in 1.13.1-beta1

in 1.13.1-beta1 don't work button "collor' and "night" =(
in sniffer packets don't send

****
Firmware:milight-hub
Version:1.13.1-beta1
IP Address:10.10.1.31
Variant:nodemcuv2
Free Heap:13584 bytes
Arduino Version:3.1.2
Last Reset Reason:Software/System restart
Dropped Packets:0

SergioLuxx, 2025-01-15:
ESP8266_d1_mini firmware 1.13.1-beta2
In the controller control card, if you switch to the color mode, it (the color mode) freezes and it is impossible to switch to any other mode, including white. From now on, the color mode remains selected.
After that, the mode is changed only through an API call