sidoh/esp8266_milight_hub #852: Color Status not updating on Home Assistant

ArshSoni, 2024-11-23:
### Describe the bug

After updating MilightHub to 1.13.0, Home Assistant no longer shows the correct brightness/color attributes on entities. 

[image]

[image]

### Steps to reproduce

### Expected behavior

### Setup information

#### Firmware version

#### Output of http://milight-hub.local/about

```json
"... /about output. put between the ```s"
```

#### Output of http://milight-hub.local/settings

```json
"... /settings output. put between the ```s"
```

### Additional context

Jojo-1000, 2024-11-23:
Hi, I am using it with home assistant on this version and the color is updating correctly. Can you check what MQTT is received?
You can subscribe to `milight/state/#` on the device settings for MQTT to see what it gets.
Mine looks like this: Topic `milight/state/0xA4A8/rgb_cct/1`
```
{
    "state": "ON",
    "brightness": 82,
    "color": {
        "r": 133,
        "g": 255,
        "b": 119
    },
    "color_mode": "rgb"
}
```

The one thing that doesn't work is if you control multiple lights at once using the controls for all.

Jojo-1000, 2024-12-02:
You should try `"color_mode": "color_temp"` instead of "white" if you have not yet. It might be that homeassistant thinks they are in ON/OFF mode, so they don't support brightness.

Then there might be something wrong with the configuration. If you go on the milight-hub device there is an advanced option "MQTT Info" which should have the internal configuration values for the lights. Mine looks like this:
<details>

```
MQTT discovery data:
Topic: homeassistant/light/milight_hub_.../fut089_0xAF77_1/config
```
```yaml
schema: json
effect: true
device_class: light
payload_not_available: disconnected
max_mireds: 370
name: LED CCT
unique_id: CFC714-LED CCT
payload_available: connected
state_topic: milight/state/0xAF77/fut089/1
min_mireds: 153
device:
  identifiers: '...'
  name: milight-hub
  model: d1_mini
  manufacturer: espressif
  sw_version: esp8266_milight_hub v1.13.0-11-g1ac8a09
  configuration_url: http://192.168.178.111
effect_list:
  - night_mode
  - white_mode
  - '0'
  - '1'
  - '2'
  - '3'
  - '4'
  - '5'
  - '6'
  - '7'
  - '8'
supported_color_modes:
  - rgb
  - color_temp
availability_topic: milight/client_status
command_topic: milight/commands/0xAF77/fut089/1
```
</details>

I would expect that your light should have `supported_color_modes` matching your devices, so the CCT only has one, which might be why it works without specifying it.

ShVerni, 2024-12-02:
Great news, I found the issue! After originally updating to`1.13.0` from version `1.11.1`, and after having originally set this up many years ago on an even earlier version, it seems the default value of the `Sate > State Fields` option was set to `Custom`, probably a holdover from a much earlier configuration setting. Changing that to `HomeAssistant` fixed the issue. Thanks for all the help!

sidoh, 2024-12-02:
Glad you figured it out! 

I adjusted the defaults in 1.13 to be HomeAssistant-compatible, but settings from older versions should carry over. If it stopped working after an upgrade that's a bug. But it sounds like you maybe just had defaults from an older version that carried over.

closed 2024-12-02 by sidoh